Ultimaker/CuraEngine #1464: Improved Print Preview, media controls and more.

Hawkeye4040, 2021-07-04:
**Application Version**
4.10.0

**Platform**
Windows 10

I would like to request the addition of improved media style controls for the preview feature of Cura. I would like to first say I understand that this is a lower priority feature, however, I didn't see it on the list so I figured I'd add it. I'd even be willing to help with it in my spare time. I have the following ideas on how the print preview feature could be improved.

- Display time on seeking and on hover over the position indicator
- Display clock with current print time and layer (option to enable/disable in preferences)
- Display clock could also have an optional indicator that estimates layer print time and shows current time/total layer time format alongside the overall print time/total print time indicator.
- Playback Speed for print preview including slow motion and fast playback controls

**Additional Information**

I personally would like to keep the print preview up and running on another monitor as I'm working on other details on my primary. I often do when doing stuff outside of Cura but having the ability to do that inside Cura would be great. I'm certain this would require the use of multiple windows which can get complicated to implement pretty fast but I'd also be willing to help with it.

This idea could really be expanded upon too and I'd really like to see that and other ideas anyone might have to enhance the print preview feature. I think it's an awesome and incredibly useful feature and I'd like to see more from it.

[image]

fvrmr, 2021-07-12:
Hi @Hawkeye4040 thank you for your feature request. 
I assume you would like this as a live print feature? If so this is a duplicate of: https://github.com/Ultimaker/Cura/issues/6020 
Let me know what you think.

Hawkeye4040, 2021-07-12:
@fvrmr it is not I think. I want the feature for print preview not monitoring live prints. I read over the linked issues and I'm pretty sure I'm following you on that.

fvrmr, 2021-07-13:
Hi @Hawkeye4040 thank you for letting me know. I will bring this up for discussion with the team. Keep you posted!

Hawkeye4040, 2021-07-13:
@fvrmr okay good so I did have that right then.

fvrmr, 2021-07-20:
Hi @Hawkeye4040 I have discussed it with the team. And we decided to defer this feature request. It would be very complex to implement time stamps into the print preview, since the time estimation is not related to the print simulation. You are welcome to take a look at it yourself off course and create a pull request.

Hawkeye4040, 2021-07-20:
@fvrmr Thank you for taking the time for me. I kind of figured this would be the case. I'm glad to see it logged on the board though. If I have the time in the near future I could get that done myself too.