Ultimaker/CuraEngine #1466: Useless supports to nowhere/hanging in the air being generated

mah115, 2021-07-12:
**Application Version**

4.10.0
**Platform**

Windows 10
**Qt**

**PyQt**

**Display Driver**

**Steps to Reproduce**

Slice attached project

**Actual Results**

Support structures that don't support anything and also hanging in mid-air being generated.
**Expected results**

Supports should actually support something
**Additional Information**

[image]
[image]
[image]
[image]
[image]
[CFFFP_Fan_Duct v8.zip](https://github.com/Ultimaker/CuraEngine/files/6803173/CFFFP_Fan_Duct.v8.zip)

fvrmr, 2021-07-16:
Hi @mah115 if you change the `support horizontal expansion` and `support interface horizontal expansion` to 0mm those towers are not printed. 
And the support that is generated from mid air, is happening because of the support pattern. A work around for this is to turn your model with the blue axis. 
Let me know if that works for you!

Ghostkeeper, 2021-07-20:
Duplicate of https://github.com/Ultimaker/Cura/issues/7487.

closed 2021-07-20 by Ghostkeeper