Ultimaker/CuraEngine #1468: bug

yangminglong, 2021-07-16:
`
void AABB3D::include(Point3 p)
{
    min.x = std::min(min.x, p.x);
    min.y = std::min(min.y, p.y);
    min.z = std::min(min.z, p.z);
    max.x = std::max(max.x, p.x);
    max.y = std::max(max.y, p.y);
    max.z = std::max(max.z, p.z);   
}

void AABB3D::include(const AABB3D& aabb)
{
    include(aabb.min);
    include(aabb.max);
}
`
repair :
`
void AABB3D::include(const AABB3D& aabb) 
{ 
	min.x = std::min(min.x, aabb.min.x); 
	min.y = std::min(min.y, aabb.min.y); 
	min.z = std::min(min.z, aabb.min.z); 
	max.x = std::max(max.x, aabb.max.x); 
	max.y = std::max(max.y, aabb.max.y); 
	max.z = std::max(max.z, aabb.max.z); 
} 
`

rburema, 2021-07-16:
This is not a bug:

`void AABB3D::include(const AABB3D& aabb)` will call `void AABB3D::include(Point3 p)`, which is possible, since both `aabb.min` and `aabb.max` are `Point3`.

It would be possible to fold these into one as you propose, but there are places in the engine where we call the `Point3` version of the function outside of the bounding-box code ( for example: https://github.com/Ultimaker/CuraEngine/blob/763c299fa8e494ddc5a75a9910208e542127ba48/src/gcodeExport.cpp#L806 ). So we need to keep the first version. Replacing the second version with the proposed one would duplicate code, so that is also less than ideal.

yangminglong, 2021-07-16:
sorry, my english is soso. I didn't describe it clearly。

void AABB3D::include(const AABB3D& aabb)
{
include(aabb.min);
include(aabb.max);
}

`AABB3D  aabb1(Point3(-5,-5,-5), Point3(5,5,5));`
`AABB3D  aabb2;`
`aabb1.include(aabb2);`

aabb1 is not  AABB3D  (Point3(-5,-5,-5), Point3(5,5,5)) now.

rburema, 2021-07-16:
I see what you mean now. That is indeed a bug. I'll reopen the issue at least.
Thanks for reporting. Since should be easy to fix, and low risk, I can probably bypass our ticket system.
I'll try to fix it later today.

rburema, 2021-07-16:
I've pushed the fix. Thanks!

closed 2021-07-16 by rburema

Ghostkeeper, 2021-07-23:
Not so sure about this. If you push a negative bounding box, should it still have a positive influence on the current AABB? I'd think that is either degenerate or just should not have any influence on the current AABB.

commit fc99b2d1 (2021-07-16): Fix for empty/negative aa-box included into another one.

Fixes #1468