Ultimaker/CuraEngine #1469: how to sort the line segments of each polygon filling line

Ccaomei, 2021-07-18:
For the filling line of each polygon, how to sort the line segments of each polygon filling line in 3D printing? Please give me some advice. Thank you

fvrmr, 2021-07-19:
Hi @Ccaomei is your question related to Cura? If so is it related to a bug you've found or a feature request?
If not please use our forum to ask questions. 
But if it's related could you give a bit more information about your question.

Ghostkeeper, 2021-07-26:
For polygonal infill patterns (like Concentric and Cross), the line segments are sorted in counter-clockwise order, starting from the vertex that is closest to the previous nozzle position before the polygon is printed.

I hope that answers your question.

closed 2021-07-26 by Ghostkeeper