Ultimaker/CuraEngine #1471: Unable to slice from the command line

albertoxamin, 2021-07-21:
**Application Version**
4.10.0

**Platform**
Mac OS 11.4 (20F71)

**Qt, PyQt**

downloaded from ultimaker website

**Display Driver**

**Steps to Reproduce**

```bash
export CURA_ENGINE_SEARCH_PATH='/Applications/Ultimaker Cura.app/Contents/Resources/resources/definitions'

'/Applications/Ultimaker Cura.app/Contents/MacOS/CuraEngine' slice -v -j '/Applications/Ultimaker Cura.app/Contents/Resources/resources/definitions/creality_base.def.json' -l models/pikachu.stl; echo $? 
```

**Actual Results**

```

Cura_SteamEngine version 4.10.0
Copyright (C) 2021 Ultimaker

This program is free software: you can redistribute it and/or modify
it under the terms of the GNU Affero General Public License as published by
the Free Software Foundation, either version 3 of the License, or
(at your option) any later version.

This program is distributed in the hope that it will be useful,
but WITHOUT ANY WARRANTY; without even the implied warranty of
MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
GNU Affero General Public License for more details.

You should have received a copy of the GNU Affero General Public License
along with this program.  If not, see <http://www.gnu.org/licenses/>.
[1]    15769 abort      '/Applications/Ultimaker Cura.app/Contents/MacOS/CuraEngine' slice -v -j  -l 
134
```

**Expected results**

I expected the program to slice the model, but instead, it aborts with exit code 134

**Additional Information**

R0B3RDV, 2021-07-22:
Maybe you should escape the spaces

albertoxamin, 2021-07-22:
@R0B3RDV no, the path is ok as there are the `'` and as you can see the program is being run, the problem is that something is causing the program to end with status code 134

fvrmr, 2021-07-22:
Hi @albertoxamin I tried to reproduce your issue by using the following commands: 
```
setx CURA_ENGINE_SEARCH-PATH "C:\Program Files\Ultimaker Cura 4.10.0\resources\definitions"

"C:\Program Files\Ultimaker Cura 4.10.0\CuraEngine.exe" slice -v -j "C:\Program Files\Ultimaker Cura 4.10.0\resources\definitions\creality_base.def.json" -l "Downloads/cube-test.stl"
```

But I couldn't reproduce your issue. Could you share your stl file?

albertoxamin, 2021-07-22:
This is the 3d model https://www.thingiverse.com/thing:376601/files
But I'm getting the issue even with different models

Ghostkeeper, 2021-07-28:
With that model I'm still not able to reproduce the problem. This is the command I'm executing:

```
CURA_ENGINE_SEARCH_PATH=/home/trin/Gedeeld/Projects/Cura/resources/definitions /home/trin/Gedeeld/Projects/CuraEngine/build-linux/CuraEngine slice -v -j /home/trin/Gedeeld/Projects/Cura/resources/definitions/creality_base.def.json -l pikachu_1gen_flowalistik.STL; echo $?
```

That produces a whole lot of g-code in stdout. The g-code ends with:

```
M84 X Y E ;Disable all steppers but Z

Print time (s): 6629
Print time (hr|min|s): 1h 50m 29s
Filament (mm^3): 9427
M82 ;absolute extrusion mode
M104 S0
;End of Gcode
```

Seems correct to me. Of course, fvrmr tested on Windows and I tested on Linux so it'll not be exactly the same.

One possibility is that you don't have permissions to execute the binary. I'd expect it to give a different OS-specific error then, but still you might want to look into that. Of course, you wouldn't listen to a guy on the internet telling you to execute their binary blob with sudo permissions... But maybe `chmod +x` needs to be done. Or maybe the signing of Cura's builds for MacOS somehow prevents you from executing individual elements of the package altogether. Should you extract CuraEngine out of the package first? (Also extract any Arcus libraries then because it links to that.)

Otherwise, you should probably run it through GDB and see where it's actually crashing. If GDB crashes as well, it's likely that it's still a permission error. If GDB doesn't crash then you can perhaps get a traceback by which we could debug the problem.

albertoxamin, 2021-07-29:
indeed it was a permission error, thanks

closed 2021-07-29 by albertoxamin