Ultimaker/CuraEngine #1472: [4.10.0] Gaps in sliced mode at wall to floor/ceiling seam

mah115, 2021-07-26:
**Application Version**

4.10.0
**Platform**

Windows 10
**Qt**

**PyQt**

**Display Driver**

**Steps to Reproduce**

Slice attached project
**Actual Results**

Holes in the sliced model exterior.
**Expected results**

No holes in the sliced model
**Additional Information**

[image]
[image]
Actual print:
[image]

Comparison to PrusaSlic3r:
[image]
[image]
[CURA_4.10_Gaps.zip](https://github.com/Ultimaker/CuraEngine/files/6880378/CURA_4.10_Gaps.zip)

fvrmr, 2021-07-30:
Hi @mah115 if you change the `top removal width` to 0 the gaps will be gone. Let me know if that solves your problem.

mah115, 2021-08-01:
> Hi @mah115 if you change the `top removal width` to 0 the gaps will be gone. Let me know if that solves your problem.

Confirm works. Thanks!

closed 2021-08-01 by mah115