Ultimaker/CuraEngine #1473: Z gap introduced slicing 100 mm cube

richfelker, 2021-07-26:
Using CuraEngine 4.9, this trivial 100 mm cube sliced at 0.2 mm layer height gets a single layer gap at Z=14.2 where most of the cross section (all but a diagonal) is missing:

[image]

[cube100.stl.txt](https://github.com/Ultimaker/CuraEngine/files/6881670/cube100.stl.txt)

This is using nothing but stock `fdmprinter.def.json` as config and setting `layer_height` and `layer_height_0` to 0.2 from the command line.

The STL file was produced by OpenSCAD and is human-readable text, and looks completely reasonable.

fvrmr, 2021-07-30:
Hi @richfelker this seems to be a duplicate of https://github.com/Ultimaker/Cura/issues/9979 we have a ticket on our backlog for this. So I will close this issue as a duplicate.

closed 2021-07-30 by fvrmr

richfelker, 2021-07-30:
OK. FWIW I think this is the best test case so far, as the geometry is utterly trivial. Please make sure whoever's looking into the ticket sees it.