Ultimaker/CuraEngine #1490: Skin Edge Support sometimes emits same infill region multiple times

richfelker, 2021-08-21:
For many files I've sliced, the Skin Edge Support feature results in the exact same infill region being emitted multiple times (possibly with different zig-zaggification). I suspect it's a logic error with how the region is split when the internal edge support lines are added, causing the same region to appear more than once (or possibly overlapping non-identical regions to get infill applied). This produces enormous nozzle backpressure, trying to print over already-printed infill lines, and quickly causes the extruder hob to start grinding filament, jamming the extruder and causing the rest of the print to fail.

I can try to provide some example cases. I know I've seen it multiple times in the past with both my own files and models from Thingiverse, and I just encountered it again with https://thangs.com/CHEP/CHEP-Universal-Building-Block-24184

smartavionics, 2021-08-22:
Please provide a project file that suffers from this issue and I will investigate. Thanks.

richfelker, 2021-08-22:
I don't use the frontend with project files but I can provide a command line with `fdmprinter.def.json` and minimal or empty options on top of that. I'll check if it's reproducible with no options.

richfelker, 2021-08-22:
I'm not sure if I can reproduce it without zig-zaggify, but this works:

```
CuraEngine slice -j fdmprinter.def.json -s zig_zaggify_infill=true -l Flexible\ Block.stl
```

The resulting file has 8 layers of skin at the top, one non-broken layer just below it, and then 3 broken layers with infill emitted twice at Z=5.2, 5.3, and 5.4.

The stl file is from the link above. I can attach it if you have problems getting it.

smartavionics, 2021-08-23:
Please provide the model, thanks.

richfelker, 2021-08-23:
Here is the file. [chep_flexible_block.stl.txt](https://github.com/Ultimaker/CuraEngine/files/7032235/chep_flexible_block.stl.txt)

smartavionics, 2021-08-23:
Thanks for the stl, could you also please provide the gcode file that you get as I can't reproduce the problem yet. Thanks.

richfelker, 2021-08-23:
[skin_edge_bug.gcode.txt](https://github.com/Ultimaker/CuraEngine/files/7032618/skin_edge_bug.gcode.txt)

smartavionics, 2021-08-23:
OK, thanks for that. Yes, I can see the duplicated infill. It doesn't happen in my Cura build because I have the following additional code which the Cura devs may wish to consider adding to their code...

```
@@ -1656,6 +1754,9 @@ bool FffGcodeWriter::processSingleLayerInfill(const SliceDataStorage& storage, L
                 infill_lines_here.cut(tool);
             }
             infill_lines.add(infill_lines_here);
+            infill_lines_here.clear();
             // normal processing for the infill that isn't below skin
             in_outline = infill_not_below_skin;
             if (density_idx == last_idx)
```

I think I have mentioned this before somewhere but my memory fails me.

smartavionics, 2021-08-23:
Ah yes, it was discussed in https://github.com/Ultimaker/CuraEngine/pull/1422 where @Ghostkeeper didn't like something so the fix never got used. Hard luck...

smartavionics, 2021-08-23:
The original fix was https://github.com/smartavionics/CuraEngine/commit/f4b96769ad29971c92fce33c9a59cbe102b45f36

richfelker, 2021-08-23:
OK, I've confirmed that this fixes the problem. Thanks! I hope the maintainers will consider merging your PR to close this.

richfelker, 2021-08-23:
FWIW this bug literally *broke my printer*, in the sense that the teeth of a 3D printed gearbox I use prior to a remote direct drive flex shaft broke after repeatedly trying to extrude against existing infill with nowhere to go. I had sufficient spare parts to replace it, and this was designed as the weakest point to avoid damage to valuable non-printed components, but I still think this makes it a severe issue meriting priority fixing.

smartavionics, 2021-08-24:
I'm truly sorry that your printer was broken by my bug. I did try to get a fix for the bug into UM Cura as soon as I discovered the problem but as you can read in the PR cited above, it was never taken up by UM.

richfelker, 2021-08-25:
No need to apologize - bugs happen. I just want it to get fixed promptly now that it's known (for the second time) so that it doesn't affect anyone else. And I'm glad to have it working now so I can turn this feature back on, because I feel like it really does help the strength of parts with protrusions on top of flat skin.

fvrmr, 2021-08-30:
We have discussed it with the team, and since it will be fixed in 5.0 we won't fix it now for releases before 5.0, I hope you understand.

richfelker, 2021-08-30:
Will 5.0 be the switch to Arachne? If so (well even if not, but especially if so) would you at least merge the PR to present mainline so that there will be a non-Arachne version without the bug, even if you're not making a release with it? I'm liking Arachne a lot now that some of the issues I hit seem to be fixed but I suspect there will be long-term use for the old engine too in some applications.

Ghostkeeper, 2021-09-01:
Yes, 5.0 is planned to be the release of the Arachne.

We provide security support for LTS releases of Cura (4.10 was LTS), but I don't think this would be classified as a security issue. We are currently maintaining these two versions of CuraEngine. Essentially, you could see the 5.0 release as the end of supporting both versions.

The refactor for the skin area generation that fixed this cannot be backported to master since it was made to deal with skin areas delimited by variable width lines. We could write a new fix, but that's precisely what we chose not to do, setting priorities based on how uncommon this problem is, how easy the workaround is and the damage of the result.

richfelker, 2021-09-01:
The one-line patch in the overlooked PR above fixes it. Why not just apply this to mainline??

ky438, 2022-12-08:
@smartavionics - has this been fixed?

smartavionics, 2022-12-08:
Hello @ky438 , sorry, I don't know the answer to that as I do not follow the Cura 5 development.

richfelker, 2022-12-11:
The fix is in Cura 5 and has been since long ago, as described above. It just didn't get a backport to Cura 4, despite it being a trivial fix to Cura 4. I had the fix in my Cura 4 branch, but I haven't used that in a long time.