Ultimaker/CuraEngine #1495: Arachne regression: inset_direction not always honored

richfelker, 2021-09-01:
One thing Arachne does really nice is, for regions that have "holes", not insisting on printing the outer-side outer wall and all the hole-side outer walls before doing inner walls when "outside_in" order is used. That is, if X is the exterior boundary and H the "hole" boundary, it prints walls in the order X0, X1, X2, H0, H1, H2 rather than X0, H0, ... (this mainly matters if there's more than one hole).

However, in the special case where only one wall fits between the exterior boundary and hole boundary, it prints X0, X1, H0, where X1 is the same as H1 would be. This means it's printing an inner wall adjacent to H0 before H0, violating the constraint that outer walls be printed first. Since X1 will be printed with variable extrusion width, the placement of its material will not be as precise, and this ends up causing the dimensional accuracy of the subsequently-printed H0 to suffer (which is the whole problem that outer_in exists to solve).

There should be some additional constraint that no inner walls of X that would touch H0 can be printed before H0 is printed.

Ghostkeeper, 2021-09-13:
I suggest you enable the Order Inner Walls By Inset setting then. This will strictly sort the walls by inset, ensuring that no outer wall is printed next to something else than an outer wall.

I'd argue that the Wall Ordering is still honoured but not sufficient to guarantee that certain walls from opposite directions are not adjacent to each other, which is quite a different - and harder - problem.

richfelker, 2021-09-13:
I have that option on already.

richfelker, 2021-09-13:
Yes, I don't believe this happened with 4.11 (or any non-Arachne Cura); that's why I labeled it a regression. I'll see if I can make a good minimal testcase.

richfelker, 2021-09-13:
[wallorder_regression.stl.txt](https://github.com/Ultimaker/CuraEngine/files/7154586/wallorder_regression.stl.txt)

richfelker, 2021-09-13:
The right angles in this test case make it particularly good. Arachne "cuts the corners" for the inner wall presumably due to varying line width considerations (center deviation mode), but since there's not already the inward-facing outer wall, the material is likely to follow a path that doesn't fully fill the corner, then the inward-facing outer wall can't be placed where it was supposed to go, resulting in an inner cavity with serious dimensional accuracy problems at the corners.

nallath, 2021-09-14:
I've added the ticket to our backlog: CURA-8571

richfelker, 2021-09-14:
FYI I went ahead and printed one of the test piece, and the rounding (and apparent gap between the corners of the outer-outer wall and inner wall) are quite prominent. The filament I had handy doesn't photograph well or I'd post a pic; I can try with a different one if anyone would like to see it.

richfelker, 2022-02-02:
Excellent! Looking forward to fix being mergable, since this is one of the big blockers for being able to replace Cura 4.x.

closed 2022-02-22 by richfelker