Ultimaker/CuraEngine #1496: Arachne regression: ironing_inset does not work

richfelker, 2021-09-02:
Using CuraEngine 4.10, ironing_inset works as expected. With Arachne, the ironing toolpath runs all the way to the edge of the surface, producing jagged edges that require significant manual cleanup with a knife or deburring tool.

richfelker, 2021-09-02:
OK, reading the source, it looks like this issue is related to the bug fixed in https://github.com/Ultimaker/CuraEngine/issues/1265. In particular, the same calculation for a width scale is used, neglecting the possibility that `infill_sparse_thickness` is less than `layer_height` (which is the default in `fdmprinter.def.json`):

https://github.com/Ultimaker/CuraEngine/blob/d7f2d305a02ef52303c85e8942ef4efef90fcf23/src/TopSurface.cpp#L67

The correction here may not even be correct in other cases, though.

richfelker, 2021-09-02:
Actually, even with that compensated for, or with the inset amount turned way up, the toolpath still goes all the way to the edge. So it looks to me like there's a bug somewhere else too.

Ghostkeeper, 2021-09-13:
https://github.com/Ultimaker/CuraEngine/pull/1272 was merged to master and it's not currently broken in Master. So I think that was only sideways related at best, but the real issue is still in the changes we made for Arachne somewhere.

Ghostkeeper, 2021-09-14:
Developers, see issue CURA-8569.

richfelker, 2022-01-07:
Ping. Is this still broken? I see a potentially related change in 95635c2897 for CURA-8615.

Ghostkeeper, 2022-01-17:
I'm afraid it is. I tested it by loading in a simple cube and setting the ironing inset to 3mm to make it visible. But the inset wasn't applied to the ironed area.

Our ticket CURA-8569 is still open. It's marked as a must-have for 5.0. The change you pointed out was to fix combing through the ironed area. The travel moves were sometimes mistakenly thinking that they were moving outside of the part instead of on top of the part.