Ultimaker/CuraEngine #1497: resource: How do I compile this engine into a dynamic library.

Nicolas8989, 2021-09-06:
I successfully compiled a static library on Ubuntu, But I can't find the release in config or makefile or the code selected to compile into a static library.

nallath, 2021-09-06:
The engine isn't intended to be compiled into a library. Why do you want to do that?

Nicolas8989, 2021-09-07:
Since we are going to call this engine on a web page, we need to compile it into a dynamic library.

Nicolas8989, 2021-09-07:
> The engine isn't intended to be compiled into a library. Why do you want to do that?

In cmakelist. TXT, change two STANTIC to SHARED.

nallath, 2021-09-07:
> Since we are going to call this engine on a web page, we need to compile it into a dynamic library.

You don't have to. You could also setup a web runner with a queue and let web page call that.

Anyhow, we only use the engine as an executable. Maybe someone else uses it as a shared library, but I can't help you with that, sorry.

Nicolas8989, 2021-09-07:
> You don't have to. You could also setup a web runner with a queue and let web page call that.
>
> Anyhow, we only use the engine as an executable. Maybe someone else uses it as a shared library, but I [...]

Thank you, I've solved this problem, but do you know what interfaces I need to call if I want to call this engine.

Ghostkeeper, 2021-09-14:
We compile CuraEngine as a library and then link that library to main.cpp as well as to our automated tests. That way we don't have to re-compile lots of stuff for every test we add.

> but do you know what interfaces I need to call if I want to call this engine.

That would depend on what you mean by "call the engine". If you compile a library, what you have is a library. The library contains classes and functions. You could call the functions and construct instances from the classes.

If you want to run CuraEngine like the executable would do, you need to call the same things that [main.cpp](https://github.com/Ultimaker/CuraEngine/blob/master/src/main.cpp) does. Basically `Application::run`.

closed 2021-09-14 by Ghostkeeper