Ultimaker/CuraEngine #1532: Bug: Extraneous Z-Hops in Special Modes -> Surface Mode -> Surface

IkonOne, 2021-11-18:
**Application Version**
Cura version: master
CuraEngine version: master

**Platform**
PopOS! 20.04 (basically ubuntu)

**Qt**
N/A

**PyQt**
N/A

**Display Driver**
N/A

**Steps to Reproduce**
1. Clone Cura, build and run.
2. Click through all of the defaults to get to a build platform.
3. Import they gyroid.stl provided in this issue.
4. In the 'Print Settings' menu, navigate to 'Special Modes' and set 'Surface Mode' to 'Surface'
5. Slice

**Actual Results**
The STL is sliced.  However, there are many extraneous z-hops.  This leads to bad TSP solutions.

[image]

**Expected results**
z-hops that occur strictly at the ends of the discretized curves.

**Additional Information**
We are 3d printing bone scaffolds using a uv cured ceramic slurry.  This slurry has a relatively low viscosity so it leaks.  In our specific use case, the z-hops lead to a lot of extra material being deposited.  And the bad TSP solutions also result in excess leaked material.

I am a CS major doing undergraduate research to support the BioMed lab at Colorado State University that is doing the actual 3d printing of bone scaffolding.  I would very much like to resolve the issue myself and will work at getting a PR ready ASAP.

-Erin

This zip file is just a renamed .3mf file.
[gyroid.zip](https://github.com/Ultimaker/CuraEngine/files/7565680/gyroid.zip)

Ghostkeeper, 2021-11-25:
The surface mode is a bit of a cop-out also for when stitching doesn't work. You'd expect these lines to get stitched together so that there is 0 distance between them even when there are rounding errors, but perhaps too much of stitching gets skipped in surface mode, causing this issue.

Ghostkeeper, 2021-11-25:
I can reproduce this problem with the following project file (not just 3MF but also including printer and settings):
[surface_mode_travels.zip](https://github.com/Ultimaker/CuraEngine/files/7603830/surface_mode_travels.zip)

IkonOne, 2021-11-26:
I think I understand the intent of stitching now.  By the documentation [here](https://support.ultimaker.com/hc/en-us/articles/360012614159-Mesh-fixes-settings), it is for filling holes in non-manifold meshes.  What I gather from your comment is that there is a side-effect that it will also fill holes caused by rounding errors that make the mesh look like it's not 'watertight' from Curas point of view.

This is good to know.  I had guessed that it was a rounding issue as well but it's good to know that you think so and that the issue is not necessarily that of the surface mode code.  Thank you.  This will probably save me a ton of digging.

IkonOne, 2021-11-26:
> You'd expect these lines to get stitched together so that there is 0 distance between them even when there are rounding errors, but perhaps too much of stitching gets skipped in surface mode, causing this issue.

Yep, you are exactly right.  The if statement here makes sure that stitching only occurs if and only if surface slicing mode is not enabled.

https://github.com/Ultimaker/CuraEngine/blob/444bc8cb3339a9be9606d40c5633880ce7a957b5/src/slicer.cpp#L751-L754

So I removed the if statement and simply stitch everything.  It works exactly as expected in that there are no extra z-hops.  However, that mode is now closing polygons.  In our use case, we need full control over that.

I don't know how I'm going to approach the resolution yet.

SS and project file for reference.

[image]

[surface_mode_closed_polygons.zip](https://github.com/Ultimaker/CuraEngine/files/7606997/surface_mode_closed_polygons.zip)

PR #1536 (2021-12-10): Fixed extra z-hops in Surface slicing mode
closes #1532 

Open poly lines were never cleaned up the same way that polygons are.
closed 2021-12-10 by rburema