Ultimaker/CuraEngine #1552: Arachne regression: Seam settings are not honored

richfelker, 2022-01-08:
As best I can tell, Arachne is ignoring the seam settings or at least not fully honoring them. On a calibration cube test print, despite my requesting seam at the back of the cube, the seam jumps between one of the back corners, one front corner, and points inside the lettering on the X and Y faces.

This was mentioned in https://github.com/Ultimaker/CuraEngine/issues/1448#issuecomment-842286592, but there doesn't seem to be a corresponding public tracker issue for the bug, so I've opened it here. I just confirmed with latest git source that the issue is still present.

Ghostkeeper, 2022-01-17:
It seems to be working correctly as far as I can tell:
[image]

This is a project file that shows it working (I know you tend to use CuraEngine without front-end, but if you also have the front-end available this is an easy way to see it): [seam_in_back.zip](https://github.com/Ultimaker/CuraEngine/files/7881291/seam_in_back.zip)

Do you have reproduction steps that show it not working?

Keep in mind that the Seam Corner Preference should still be held. If the preference says to hide the seam, it will only select from the inner corners, unless there aren't any. Also keep in mind, especially when calling the Engine without front-end, that the actual settings being used are the `z_seam_x` and `z_seam_y` settings. The `z_seam_position` setting is just a convenience for the front-end that automatically sets the X/Y position to a certain corner of the build plate.

richfelker, 2022-02-02:
I missed that you'd responded to this. I'm trying latest with my test case with latest git, since I was still on a revision well older than your reply. It's a simple calibration cube, nothing fancy, where I want the seam at a single corner all the way up, not jumping around with geometry. I'll let you know how it comes out and take a look at your test case too if I'm still hitting problems.

richfelker, 2022-02-02:
OK, it still happens with my test case file, 
[mycalcube.stl.txt](https://github.com/Ultimaker/CuraEngine/files/7990479/mycalcube.stl.txt)

Printed with base `fdmprinter.def.json` plus:

```
-s z_seam_type=back \
-s z_seam_x=0 \
-s z_seam_y=100 \
-s infill_line_distance=0 \
-s layer_height_0=0.2 \
-s layer_height=0.2
```

At layer height 2.0 and 2.2 it moves to the front left corner rather than the back left corner to start. In my actual profile it jumps around at a few other heights to, but I reduced this to a minimum test case for reporting.

richfelker, 2022-02-17:
Perfect! I actually have that PR branch checked out, and indeed it does fix my test case - not just the minified settings version of it above, but with my actual profile too.

richfelker, 2022-02-22:
And with #1562 merged, this is fixed. Closing now. Thanks!