Ultimaker/CuraEngine #1557: Issue while installing with cmake . .

floxform, 2022-01-18:
Hello!
While installing the curaengine on Ubuntu I ran into some errors. 
This is the error message, if I try to run cmake . . 
I'm in the directory: CuraEngine-master/build/

Thank you in advance !

//// ERROR MESSAGE
-- Building with Arcus
CMake Warning (dev) at /usr/local/lib/cmake/Arcus/ArcusConfig.cmake:30 (set):
  implicitly converting '' to 'STRING' type.
Call Stack (most recent call first):
  CMakeLists.txt:23 (find_package)
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Building Stb from source.
-- Created Stb installation at: /home/user/Downloads/CuraEngine-master/build/stb-prefix/src
CMake Error at /usr/share/cmake-3.16/Modules/FindPackageHandleStandardArgs.cmake:146 (message):
  Could NOT find Boost (missing: Boost_INCLUDE_DIR)
Call Stack (most recent call first):
  /usr/share/cmake-3.16/Modules/FindPackageHandleStandardArgs.cmake:393 (_FPHSA_FAILURE_MESSAGE)
  /usr/share/cmake-3.16/Modules/FindBoost.cmake:2179 (find_package_handle_standard_args)
  CMakeLists.txt:36 (find_package)

-- Configuring incomplete, errors occurred!
See also "/home/user/Downloads/CuraEngine-master/build/CMakeFiles/CMakeOutput.log".
See also "/home/user/Downloads/CuraEngine-master/build/CMakeFiles/CMakeError.log".

jellespijker, 2022-01-19:
You're missing the dependency Boost https://www.boost.org/ which is needed since we merged the Arachne branch into master.
Install boost and set the Boost root folder, such that CMake can find it.

I will close this issue now but feel free to reach out if building from source still fails.

closed 2022-01-19 by jellespijker

floxform, 2022-01-19:
How do I set the Boost root folder?

floxform, 2022-01-19:
It works now, after reinstall.
Thank you!