Ultimaker/CuraEngine #1561: Surface Mode - zero distance travel moves

IkonOne, 2022-01-21:
**Application Version**
4.13.0

**Platform**
All

**Qt**
N/A

**PyQt**
N/A

**Display Driver**
N/A

**Steps to Reproduce**
Slice the included stl with surface mode set as 'Surface' and slice at 0.2 layer height.

**Actual Results**
[image]

Extra travel moves on open polylines.

**Expected results**
No extra zero distance travel moves.

**Additional Information**
This commit re-introduced the issue: https://github.com/Ultimaker/CuraEngine/commit/2d07181ab54dd802f34dc3002ee8578e992b5939#

Per the description in the aforementioned commit, the simplify function is intended for closed polygons but was being used on open polylines.

I think the right fix would be to create a function for joining open polylines.

-Erin

Ghostkeeper, 2022-01-28:
We have simplify adjustments in progress to handle polylines better: https://github.com/Ultimaker/CuraEngine/pull/1554

It is a necessity in the Cura 5.0 release due to the wall generation adding polylines.
This should also make it possible to fix this issue again.

IkonOne, 2022-01-28:
Great.

For anyone who needs continuous surface mode lines, try Cura version 3.14
Beta. That one works for us since we are printing non-manifold geometry.

closed 2026-10-26 by IkonOne