Ultimaker/CuraEngine #1573: Feature request: improved min infill area

richfelker, 2022-02-07:
Use of "area" as the criterion for minimum infill area admits narrow, complex- or awkward-shaped small regions to get processed as infill, which can harm print time and quality. I can imagine a better way to avoid generating infill in "small" regions doing something comparable to skin preshrink/expand:

- Given initial infill region, apply a pair of shrink/expand offset operations to "simplify" it.
- Subtract the simplified infill region from the original infill region.
- Process the difference as skin, or ideally as all-perimeters printed outward-in like "extra skin wall count" does.

For an example of where this would be beneficial, see the part of the Benchy cabin where the slices are 4 small components. Folks trying to print fast benchies generally use larger line width *just to avoid* getting infill here (and possibly in a few other places). I can think of other scenarios though where it's just a matter of the region being *awkward* and incurring a lot of zigs and zags to print a very tiny amount of actual infill line material.

I would call the improved option something like "min infill region width" since that's what the shrink/expand operation essentially enforces.

richfelker, 2022-02-19:
I've whipped up a demo implementation using OpenSCAD CSG: [min_infill.scad.txt](https://github.com/Ultimaker/CuraEngine/files/8103166/min_infill.scad.txt). Here's an image of what the demo file looks like:

[image]

The blue is the offset-and-difference computed wall region. The green and red together are the offset-computed infill region. Green is the first bullet point from the description above and red is the second.

You can test the concept on other regions just by redefining `region()` in the demo file.

Here, infill would only be generated in the green region. The red part that was initially infill but "too small for infill" would be filled either with skin or (ideally) with Arachne polylines.

Ghostkeeper, 2022-02-26:
One problem with this is that the infill would get replaced with skin, which then has the same problem of generating a bunch of tiny line segments too.

When the infill is thinner than 1 line width and Connect Infill Lines is enabled, we currently already replace that with a variable-width line. This is necessary because the connection between infill lines is a simple 1-line-width infill line which would be wider than the infill area then. We could trigger that behaviour also if infill lines are not being connected, but it has the risk that the user gets a 100%-filled part where he explicitly wants it to have lower fill rate. The 100%-fill would print better and faster, but could also cause banding. Still, worth considering, since for the user such a change would not be illogical and not add any complexity to the configuration.

The main purpose of Minimum Infill Area, in my eyes, is to prevent a flow change when it's really unnecessary. If there is other infill in a wider segment of the same part, you'd still get a flow switch. But your suggestion is to replace the infill with a different pattern in thin parts. These are applications with different constraints.

richfelker, 2022-02-26:
> One problem with this is that the infill would get replaced with skin, which then has the same problem of generating a bunch of tiny line segments too.
>
> When the infill is thinner than 1 line width [...]

The main problem occurs when the width is slightly over 1 line width but less than 3. In that case, zigzags together with the actual infill lines generates can exceed the total area they're supposed to fill, giving dimensional errors or causing the nozzle to collide on the next layer. I don't have a test case handy at the moment but I've seen some really ugly triangles/cubic paths in small oddly-shaped regions. Whether these are worse for speed than skin fill, I don't know, but they're almost surely worse for print quality. And if Cura used Arachne variable width paths to fill rather than skin fill, it'd surely be a win for print time too.

> The main purpose of Minimum Infill Area, in my eyes, is to prevent a flow change when it's really unnecessary. If there is other infill in a wider segment of the same part, you'd still get a flow [...]

Is that because it's already doing overlapping wall compensation (or now, with Arachne, variable width paths) on infill? I don't think that's the case but I may be mistaken, based on how infill self-intersection seems to be handled for triangles, cubic, etc.

In my view, he main purpose of min infill area is to ensure there are not fragile parts due to the infill pattern's gaps happening to align with the entire interior of the region. But filling the area with skin instead, you ensure it has enough surface area in contact with the adjacent layers that the part doesn't easily break. (Note: extra walls could accomplish this too, but would waste a lot of time and material on larger regions that are plenty strong without more walls.)

Ghostkeeper, 2022-03-07:
> Is that because it's already doing overlapping wall compensation (or now, with Arachne, variable width paths) on infill? I don't think that's the case but I may be mistaken, based on how infill self-intersection seems to be handled for triangles, cubic, etc.

Cura 4.x applied overlap compensation only on walls, not infill. And in 5.0 the variable line width being applied to walls and the extra infill/skin/support walls. It doesn't apply to the infill pattern itself. What I meant though is that the infill has a different flow rate because it usually prints faster and sometimes even has a much greater layer thickness. It gets printed separately, so having a small piece of infill it would have to come back for that bit later, leaving scars, taking time, reducing strength.

richfelker, 2022-03-07:
> And in 5.0 the variable line width being applied to walls and the extra infill/skin/support walls. It doesn't apply to the infill pattern itself.

That makes the need for this all the more important, since it necessarily *will* over-fill narrow regions.

> infill has a different flow rate because it usually prints faster and sometimes even has a much greater layer thickness

The region converted to skin or outlines would print with those settings in sequence with other lines of the same type and not use sparse infill thickness (multi-layer height). So I don't see it being a problem, no?

Ghostkeeper, 2022-03-20:
Not a problem if you don't mind those extra travels or that it's filling that thin part 100% instead of the selected infill density. It's all a big balancing act, of course. We currently don't have this width as a selectable setting because we think it's too much of a detail for most people to bother adjusting it (and adding a setting makes Cura more complex to use), and if it's adjusted without understanding it can ruin the print. 1 line width is normally fine.

Why is it over-filling narrow regions, you think? Or do you mean that the infill rate is higher than the selected infill density if it creates a single variable-width line? I don't think I'm following you, here.

richfelker, 2022-03-20:
> Why is it over-filling narrow regions, you think?

Self-intersecting infill patterns (grid, triangles, cubic, etc.) do not account for the double or triple coverage of the intersection points. In large regions you have plenty of space around the intersection point for the material to spill over into, but in tiny regions, it's possible that the intersection is basically all you have (depending on how it aligns with the part).

Also, zig-zaggification does not seem to account for overlap with infill lines. Unlike the self-intersecting pattern case which only overextrudes rarely when you get unlucky, *most* very narrow infill regions have largish overlap between the infill pattern and the long connecting lines between them. In a worst case, I think you can even get long connecting lines along the opposite long walls of a narrow rectangle that's barely wide enough for one line, such that there's overextrusion before you even count the infill pattern. I'm a little bit confused why this happens because I thought Cura did different insetting with zig-zaggify enabled to avoid this, but you can definitely see it, such as here in my extrusion of a 2.3 by 49 mm rectangle:

[image]

Ghostkeeper, 2022-04-10:
Yeah, the infill might be too dense or too sparse depending on where the infill lines end up. On average it should be the correct density, but locally it will indeed be wrong.

> I'm a little bit confused why this happens because I thought Cura did different insetting with zig-zaggify enabled to avoid this,

Cura needs to calculate the centreline of the infill path in the end. This centreline is half a line width from the edge of the infill area. So Cura makes an inset of half a line width from the edge of the infill area when zigzaggifying, so that it can follow the border of this inset with the connecting lines, and so that the actual infill pattern stops before the edge to match up with these connecting lines. That is what I'm seeing in your screenshot. But indeed it makes no effort to prevent partial overlap between different lines, like what you see with the N-shaped bit in the middle of your diagram there. This could be problematic if the infill line is almost exactly aligned with the shape of the thin infill part. There's no "wall overlap compensation" for infill or anything.

richfelker, 2022-04-26:
While this doesn't get skin to replace it, I'm successfully using the following patch to avoid self-intersecting infill of the above form:

```
diff --git a/src/infill.cpp b/src/infill.cpp
index 81794b462..f298ef6d2 100644
--- a/src/infill.cpp
+++ b/src/infill.cpp
@@ -130,6 +130,8 @@ void Infill::generate(std::vector<VariableWidthLines>& toolpaths, Polygons& resu
         // Now do the actual inset, to make place for the extra 'zig-zagify' lines:
         inner_contour = inner_contour.difference(gap_filled_areas).offset(-infill_line_width / 2);
     }
+    double preshrink_expand = settings.get<coord_t>("infill_preshrink_expand");
+    inner_contour = inner_contour.offset(-preshrink_expand).offset(preshrink_expand);
     inner_contour.simplify(max_resolution, max_deviation);
 
     if (infill_multiplier > 1)
```

The intended default for `infill_preshrink_expand` is half `infill_line_width` but I made it configurable at least for testing. Indeed it avoids the problem and seems to significantly improve print times in models with small details where the infill region is broken up badly between wall geometry. So I think this would be a feature users might like.