Ultimaker/CuraEngine #1574: ooze shield thickness wall option OR purge tower only in right spot

StefanoV1989, 2022-02-08:
Hi,
i wonder if it's possible to add the "ooze shield thickness wall" as an option for dual extruders printer.
I have a Cr10 Mini modified with a bigtreetech mixing extruder, and when i print i have only two "useless" options.
1. Using purge tower, even if i have just a little spot with second color on the top of the model, the purge tower is using both colors starting from bottom. It's useless and a waste of material! Just print on the tower with the actual color without changing tool for each layer!
2. Using ooze shield, i have second color only in the right spot but i can't change the number of walls for the ooze shield, so i have a mixed color because nozzle is still dirty from the previous color.
I avoided a little bit with the option "internal layer before external" but it's not enough. With 2 or 3 walls on the ooze shield, it would be perfect.

I prefer the ooze shield, but i need a wall number option to use it.

**Application Version**
4.13

**Platform**
Windows

**Display Driver**

**Steps to Reproduce**
Dual extruder printer

fvrmr, 2022-02-18:
Hi @StefanoV1989 thank you for your request.
Why do you want multiple ooze shield walls? Because you have mixed color in the ooze shield or model?
As you can read in the settings guide: 

> The prime tower consists of an outer shell which is completely printed with one extruder, every layer. This is necessary in order to guarantee stability of the tower, but will sometimes require extra extruder switches for some prints.

Read here more: https://github.com/Ghostkeeper/SettingsGuide/blob/master/resources/articles/dual/prime_tower_enable.md

StefanoV1989, 2022-02-20:
> Hi @StefanoV1989 thank you for your request. Why do you want multiple ooze shield walls? Because you have mixed color in the ooze shield or model? In the settings guide you can read more why the prime tower needs an outer shell with one extruder:
> 
> > The prime tower consists of an outer shell which is completely printed with one extruder, every layer. This is necessary in order to guarantee stability of the tower, but will sometimes require extra extruder switches for some prints.
> 
> Read here more: https://github.com/Ghostkeeper/SettingsGuide/blob/master/resources/articles/dual/prime_tower_enable.md

Hi, thank you for the answer.
I think i found a bug then.

I have this random object:
[http://www.sv-photography.org/1.png](http://www.sv-photography.org/1.png)

I select ext 1 for cylinder, and ext 2 for the cube, and everything goes ok (until it's blue, the prime tower is blue):

[http://www.sv-photography.org/2.png](http://www.sv-photography.org/2.png)

If i switch the extruders (cylinder ext2 and cube ext1), that's what happens:
[http://www.sv-photography.org/3.png](http://www.sv-photography.org/3.png)

Basically the prime tower uses both color even if it doesn't need it!

Because of that i used the ooze shield without prime tower, and i was asking for a thicker shield because only one layer is not enough to clean the single nozzle.