Ultimaker/CuraEngine #1577: Arachne regression: Vase mode bottom layers are all concentric

richfelker, 2022-02-11:
Unlike Cura 4.x, which alternated between lines and part-concentric/part-lines for bottom skin, Arachne is printing entirely with concentric skin fill. This makes it a lot harder to get a water-tight bottom, as any slight gaps between extrusions will tend to line up exactly from one layer to the next, rather than getting filled by the opposite-direction lines weave in the next layer. As far as I can tell this behavior is not customizable now or before, but I don't understand why that is; it seems like honoring the user's preferences for bottom layers would be easy to do and much better. But the specific thing I'm reporting here is the regression.

fvrmr, 2022-02-18:
Hi @richfelker thank you for your report.
We have already a ticket about this on our backlog. I will link your issue to that ticket.
Devs see CURA-7926

lukk555, 2022-07-07:
Just want to mention that on Cura 5.1 Beta this issue is still visible - no matter of top/bottom pattern and bottom pattern initial layer, the print in vase mode is always sliced with concetric bottom. Quite annoying, need to use 4.13 for vase mode prints.