Ultimaker/CuraEngine #1586: [master] How to Dump Arcus command between Cura and CuraEngine?

jeremysalwen, 2022-02-18:
**Application Version**
master

**Platform**
linux

When developing for Cura and CuraEngine, it is important to understand the information that passes from Cura to CuraEngine.  This communication uses [Arcus](https://github.com/Ultimaker/libArcus) as the protocol.  However, I don't see any way to dump this information to a file. Is such a thing possible?  Ideally it would also be possible to read this dump to invoke CuraEngine, so we could reproducibly run the slicer.

This question is somewhat related to https://github.com/Ultimaker/CuraEngine/issues/506.

jeremysalwen, 2022-02-18:
It is for understanding better the API between the two programs.  I was also thinking that it might provide an opportunity to debug the CuraEngine in a reproducible environment without having to run Cura.

I was able to get a dump of the message as textproto using the following code: https://github.com/jeremysalwen/CuraEngine/blob/master/src/communication/ArcusCommunication.cpp#L495-L499  I could also see how it wouldn't be too hard to read such a message back and feed it into CuraEngine, although based on what you are saying, it doesn't sound like that's something that would be helpful to most Cura developers.

nallath, 2022-02-18:
> I was also thinking that it might provide an opportunity to debug the CuraEngine in a reproducible environment without having to run Cura.

We tend to just use the CLI of CuraEngine in that case.

jeremysalwen, 2022-02-18:
Hmm, my understanding from https://github.com/Ultimaker/CuraEngine/issues/506 was that the configuration of MeshGroups and Meshes was not supported in CLI usage of CuraEngine, which means the only way to test them is through the GUI?

Ghostkeeper, 2022-03-07:
> Hmm, my understanding from #506 was that the configuration of MeshGroups and Meshes was not supported in CLI usage of CuraEngine, which means the only way to test them is through the GUI?

You can set settings specific to meshes by adding the `-s` parameter after the loading of the mesh. The order matters. And you can define a new mesh group with the `-g` parameter.

For almost all of my debug sessions in CuraEngine, I'll run Cura with the `--external-backend` parameter and then debug CuraEngine with GDB in my IDE. This is a more reliable reproduction environment, because Cura's front-end can load in Cura project files. For a user or a tester, these project files are much easier to create than text dumps of the Protobuf messages, so this is a better way to get a reproducible test case, even though it is equally reliable as a reproduction method once the file exists. And they would be easier to modify, too.

jeremysalwen, 2022-03-07:
Thanks for the tip to use `--external-backend`.  

I was thinking that using a proto dump would more cleanly separate the Cura testing from the CuraEngine testing. Especially if you wanted to do automated regression testing as part of CI, to check that some PR generates the exact same G-Code given the same input.  You can cleanly separate that into two stages: Project => Arcus command, and Arcus command => G-Code, and you could automatically generate a report of the diff between the proto output.

Of course, as I said before, the point is not to second guess how you do development, so if this is not interesting to you then probably it's not worth it.

Ghostkeeper, 2022-03-20:
In our CI we circumvent the communication step altogether and can directly create a Slice object, fill it with models programmatically and then start a slice that way :) This is basically what the command line mode does too. I think that this is indeed not of interest for Ultimaker as part of our workflow, so I'll close this issue.

closed 2022-03-20 by Ghostkeeper