Ultimaker/CuraEngine #1587: Meshgroup-Level settings are unused by Cura?

jeremysalwen, 2022-02-18:
**Application Version**
master

**Platform**
linux

After dumping the Slice message that Cura sends to CuraEngine, I looked at which settings in the UI correspond to which fields of the proto.  It seems that Per-Object level settings in the UI always correspond to the object-level settings in the proto, and global settings always correspond to the global_settings in the proto.   

However, I couldn't find any settings that corresponded to the MeshGroup (i.e. `object_list`) level settings in the proto.  I tried every combination of options I could, using One-at-a-time printing enabled, and disabled, grouping, and merging various objects.  None of the settings I changed showed up as `object_list` level settings in the proto, with the sole exception of just the first MeshGroup getting "extruder_nr" set to 0.

Is this expected behavior? i.e. are the MeshGroup level settings basically unused by CuraEngine, or am I missing some options in the Cura UI?

Ghostkeeper, 2022-03-07:
Indeed, this is expected at the moment. Although it works slightly differently than what you're writing.

In CuraEngine, the concept of a "mesh group" is a group of meshes that gets printed at the same time, from the first layer to the last. After printing the first mesh group, CuraEngine will restart the slice entirely for the second mesh group, back from the first layer. The front-end basically just uses this by sending each mesh as a separate mesh group when printing in one-at-a-time mode.

Because basically the entire slice is re-started, almost all settings are read at the mesh-group level or lower in CuraEngine. For instance, the Build Plate Adhesion (`adhesion_type`) setting is global according to the front-end, but CuraEngine actually requests it as a mesh group setting. So in theory you could have different settings per mesh group. You could in theory set a global setting like the adhesion type differently for each mesh when printing in one-at-a-time mode.

However in practice this doesn't work well. The extruder-level settings interfere. Cura's front-end also evaluates each setting for each extruder and sends those over to the back-end, which causes the extruder-level setting to override the meshgroup-level one. So when the ABS profile says it would prefer to use a brim, it'll use this instead of the mesh group setting you specified. 

See also https://github.com/Ultimaker/CuraEngine/blob/master/docs/settings.md , although this story isn't really explained there since it's more about the front-end than the back-end.

Let me know if this needs more clarification.

closed 2022-03-07 by Ghostkeeper

jeremysalwen, 2022-03-07:
I think I must be missing context here.

1.
>  Cura's front-end also evaluates each setting for each extruder and sends those over to the back-end, which causes the extruder-level setting to override the meshgroup-level one. 

My understanding of what you wrote is that you are saying that Cura *is* sending mesh-group level settings to CuraEngine, but since it is *also* sending extruder-level settings, CuraEngine effectively ignores the mesh-group level settings.  However, my understanding of the code is that CuraEngine never sends mesh-group level settings, and only sends mesh level settings, global settings, and extruder settings.  Which one of these understandings is wrong?

2. You seem to be implying that this is a temporary state of affairs.  Do we expect things to change soon?  What is the goal state?

Ghostkeeper, 2022-03-20:
> My understanding of what you wrote is that you are saying that Cura is sending mesh-group level settings to CuraEngine, but since it is also sending extruder-level settings, CuraEngine effectively [...]

Both, at the moment, but for different reasons.

Cura no longer sends any meshgroup-level settings to CuraEngine. This was something that was changed briefly a couple of versions ago, but was changed back because of the aforementioned problems.

CuraEngine never sends any settings anywhere, but consumes them. CuraEngine consumes settings from different levels. For instance, these skirt/brim settings are obtained here from an extruder:

https://github.com/Ultimaker/CuraEngine/blob/58cae8395a97879f36e960d8f551089c2ec394a5/src/FffGcodeWriter.cpp#L379

while this one is obtained from a mesh group:

https://github.com/Ultimaker/CuraEngine/blob/58cae8395a97879f36e960d8f551089c2ec394a5/src/FffGcodeWriter.cpp#L383

CuraEngine does read settings from mesh groups, but since the mesh group is currently always an empty settings object, it will always fall back to the global settings (in `Scene`).

> You seem to be implying that this is a temporary state of affairs. Do we expect things to change soon? What is the goal state?

I think the ideal goal state would be that you can select different settings per mesh group if you are printing in one-at-a-time mode, and that more or less all settings can work that way, even those that are changed by the extruder. And that this way, you can change settings that are normally not changeable per object, such as the printing temperature or the layer height. Most settings can be changed this way (perhaps with the exception of machine settings and stuff like Print Sequence itself). I don't expect this to change soon though, because using different settings in one-at-a-time mode is a really niche use that will not get sufficient priority.

jeremysalwen, 2022-03-20:
> Cura no longer sends any meshgroup-level settings to CuraEngine. This was something that was changed briefly a couple of versions ago, but was changed back because of the aforementioned problems.

As I understand it, there is no option in the Cura UI to change mesh-group level settings, only global settings and mesh level settings.  Was there also a UI change?  Or did you make the per-object settings become mesh-group level settings when one-at-a-time mode was enabled?

Ghostkeeper, 2022-04-10:
The meshgroup-level settings were shown in the per-object setting dialogue for a while, if the user enabled one-at-a-time mode.

This was the change: https://github.com/Ultimaker/Cura/pull/3154 Gosh, 2018 already... To be clear, this has since been reverted. We're talking about something that was temporarily a thing for 9 months in 2018.