Ultimaker/CuraEngine #1589: Cura does not lower bed to avoid highest printed part before homing X/Y at completion of One At A Time sequence print

ansonl, 2022-02-19:
**Application Version**

4.13.1

**Platform**

Windows 10

**Qt**

**PyQt**

**Display Driver**

N/A

**Steps to Reproduce**

1. Place a tall (lower left)  and short object (upper right) on the build surface in Cura. 
2. Set printing Sequence to One At A Time
3. Print

**Actual Results**

At the end of the print, when the gcode homes X/Y, Cura does not generate gcode to lower the printing bed to the tallest object printed height so the printhead will home X/Y at the height of the last object printed which may not be the tallest object printed on the build surface. The print head will slam into the already printed objects and homing will fail. 

**Expected results**

Cura should lower the bed to the tallest height of all printed objects before moving onto the next printed object in the sequence and before homing X/Y at the conclusion of the print. 

**Additional Information**
[image]

[image]
[image]

fvrmr, 2022-02-25:
Hi @ansonl thank you for your report.
Could you share your project file (file - save project) here as a .zip?

ansonl, 2022-02-25:
@fvrmr, attached.

I don't have the original project file but was able to replicate the position and sequence with two sample objects. 
When I generate gcode for the attached project, the taller object on the left prints first. When the second lower object is done printing, the printer will proceed directly to homing X/Y and will hit the taller object as it homes X first. 

[Cura_test_sequence.zip](https://github.com/Ultimaker/CuraEngine/files/8142313/Cura_test_sequence.zip)

Ghostkeeper, 2022-03-07:
This is a problem with settings, but more specifically it's a problem with the default end g-code of the Ultimaker Original+. It has this line:

    G28 X0 Y0 ;move X/Y to min endstops, so the head is out of the way

In the case of one-at-a-time printing, this is dangerous.

ansonl, 2022-03-07:
@Ghostkeeper If the printer doesn't move the nozzle out of the way at the end of the print, won't that create a risk of oozing depending on how the user's printer is set up?

Ghostkeeper, 2022-03-12:
Yes, it should move the head away before signalling end-of-print, because the end-of-print takes a while to process, retracts the filament, etc. Directly moving vertically tends to leave a little tail on the top of the print though. I believe that with the Ultimaker 2 we move horizontally by 20mm or so in the end g-code, and then end the print, after which the firmware moves the bed down. This is theoretically still dangerous (other objects could be hit by that 20mm move, or it could go out of bounds) but that almost never happens™.

richfelker, 2022-03-21:
My end gcode only moves 2 mm horizontally to break the string after retracting, and works fine. This would almost surely be safe for arbitrary one-at-a-time prints.