Ultimaker/CuraEngine #1603: Set up fuzz testing for CuraEngine

artem-smotrakov, 2022-03-12:
Fuzzing is a well-known testing technique that can help identify security issues such as overflows, use-after-free, etc.

I'd like to set up fuzz testing for `CuraEngine` by adding it to [OSS-Fuzz](https://google.github.io/oss-fuzz/) .

To begin, I need an email address for the engineering contact to be CCed on new issues, satisfying the these [requirements](https://google.github.io/oss-fuzz/getting-started/accepting-new-projects/). Unfortunately, my email address won't be accpted because I am not a maintainer. Once I have the contact, I'll add `CuraEngine` to OSS-Fuzz, look into findings and fix them.

Let me know if you have questions.

nallath, 2022-03-16:
You can add my email address! j.vankessel@ultimaker.com should match all the requirements.

rburema, 2022-11-18:
Hi! Thanks for (the process of) setting this up. I see the required action for this issue has been taken, and other follow-ups will likely be different issues. As such, I'm closing this one.

closed 2022-11-18 by rburema