Ultimaker/CuraEngine #1604: Staggered Top/Bottom

BagelOrb, 2022-03-14:
Where two top/bottom lines next to each other touch they are presumable less strong and in the middle of either line.
When alternating the top/bottom directions between 0* and 90* the less strong parts are all aligned in vertical columns.

If we shift the top/bottom pattern by half a line width for `(layer_nr % 4) / 2 = 0` then those less strong columns are broken up.

Staggered Top/Bottom is therefore expected to increase the ultimate strength of printed parts, while the computational costs and printing costs don't increase.

See the video at https://dl.acm.org/doi/10.1145/3485114.3485117
[image]
Note that this would be a diagonal cross-section at 45* to the lines at even layers and -45* the lines at odd layers.

Ghostkeeper, 2022-04-02:
The linked video talks about the defect caused by splitting and joining lines being aligned vertically. This is not an issue for the top/bottom pattern because it doesn't split/join lines. It is a problem for walls (and extra skin walls) but there shifting them by half a line width is not an option, or it would not match the outline. A similar problem occurs also when two lines are adjacent without splitting of course.

I get how it applies to stacking lines in the same direction, as in the video, because then the line on top can fill the cavity between two adjacent lines in the layer below. But why doesn't that get negated by the alternating of the line pattern direction? There's quite a lot of material to spread the forces around then.

BagelOrb, 2022-04-02:
When you consider one layer of skin then in between the extrusion lines there are lines along which less material is deposited. When considering two layers then those lines intersect at a regular square grid of locations where both layers have less material deposited. When considering more layers then those locations along the grid are stack on top of each other, which would hypothetically be more susceptible to crack lines forming.

When we offset every other two layers by half the line width then the underextrusion locations don't align in a vertical column.