Ultimaker/CuraEngine #1606: Exclusive slicing_tolerance produces empty first layer

richfelker, 2022-03-15:
Setting `slicing_tolerance` to `"exclusive"` usually (always? unless `mesh_position_z` < 0?) produces an empty first layer which gets removed. This both prevents `initial_bottom_layers` from getting used correctly, and produces a print that's one layer too short in the Z direction. The culprit seems to be `slicer.cpp` lines 857-863:

https://github.com/Ultimaker/CuraEngine/blob/b5c84d0d368df7d4a96387d5aa37ca9fb3ae8a23/src/slicer.cpp#L857-L863

whereby points exactly on the slice boundary get moved one unit above it. I suspect this should be changed to move them one unit below it, at least on the first layer.

Lowering the model by 0.01 or so works around the issue, but ends up making the print one layer short at the top (assuming the entire model height was an integral number of layers to begin with).

Ghostkeeper, 2022-04-02:
That code snippet indeed looks quite like a band-aid rather than an actual fix. For context, the linked Jira ticket (CURA-8614) refers to this issue: https://old.reddit.com/r/Cura/comments/povb83/cura_411_does_not_slice_bed_leveling_files_anymore/hd3h73n/ . It was added by this PR: https://github.com/Ultimaker/CuraEngine/pull/1516

I would expect that a more proper fix involves adjusting the `<`, `<=`, `>` and `>=` operators in the case below it.

That said though, it is expected that Exclusive tolerance makes the vertical dimension of the model smaller. That is the whole point of it, really. It's still a problem that this is taken from the lowest layer if it affects the initial layers settings. But that may need to be fixed in other ways.

rburema, 2022-11-18:
Hi! Due to a communication mixup, a lot of the engine issues got ignored lately, sorry.

I think this can be fixed (a bit hacky, but so is the original solution), by modifying the code with the above PR.

I'll bring it up with my colleagues.

PR #1776 (2022-12-30): [CURA-9921] Correct bottom-layer(s) behaviour for exclusive tolerance.
Since all points where increased by one, this lead the exclusive tolerance, in cases where the intial layer height is smaller or equal to the 'normal' layer height, to be missing a layer, since the model would be on z=0. Since the initial layer is not on the build-plate, this causes the number of bottom layers to not apply anymore.

should fix #1606
commit 9d30c843 (2022-12-30): Correct bottom-layer(s) behaviour for exclusive tolerance.

Since all points where increased by one, this lead the exclusive tolerance, in cases where the intial layer height is smaller or equal to the 'normal' layer height, to be missing a layer, since the model would be on z=0. Since the initial layer is not on the build-plate, this causes the number of bottom layers to not apply anymore.

should fix #1606
closed 2022-12-30 by jellespijker