Ultimaker/CuraEngine #1611: Travel between very close but non-connected regions get combed

richfelker, 2022-03-16:
I first noticed this a year or two ago printing screw threads oriented in the XY plane: the top one or two layers would get severe stringing between them, and seemed to be missing retractions. Setting a low `retraction_combing_max_distance` fixed it, so the problem was clearly that combing was being applied despite the travel being between disconnected regions.

I've finally tracked the bug down to:

https://github.com/Ultimaker/CuraEngine/blob/58cae8395a97879f36e960d8f551089c2ec394a5/src/LayerPlan.cpp#L384-L389

where `Comb::calc` simply returns true (do combing) if the distance is less than `max_distance_ignored`. 

I've since found that the same problem occurs in some articulated print-in-place models where it makes it harder to break them free initially and weakens small linkages by not getting the full amount of material bonded in the intended places.

As a workaround, the `machine_nozzle_tip_outer_diameter` setting can just be set to 0, but I believe this should really be fixed. Retracted travel without any stringing over distances shorter than the nozzle outer diameter *is possible*, and failure to do so is very surprising and compromised the integrity of prints.

Ghostkeeper, 2022-04-10:
A similar issue was fixed here, a few days after you reported this: https://github.com/Ultimaker/CuraEngine/pull/1619

Does this issue still occur in master? How would we reproduce it?

richfelker, 2022-04-10:
Reading those changes, it looks like they may mitigate the specific case I hit, but I think an issue is still present. `Comb::calc` unconditionally says "yes, combing is possible" when the distance is less than `max_distance_ignored`, and this is just wrong. As far as I can tell the commits don't change that because the logic for `max_distance_ignored` short-circuits the whole rest of the function's logic for when combing is allowed.

richfelker, 2022-04-12:
At the moment I don't seem to have a test case to trigger the problem, but it may also be a matter of other local changes I've made. I'll try to follow up with an unmodified tree later and see if I can find cases still affected.

rburema, 2022-11-18:
>  I'll try to follow up with an unmodified tree later and see if I can find cases still affected.

Sorry for the wait. Did you find anything?