Ultimaker/CuraEngine #1631: Small feature size speed is applied to Arachne inner-wall loops

richfelker, 2022-03-31:
In narrow regions where width varies such that short loops of inner-wall get produced, the small feature size speed limits get applied to these loops. This makes the print extremely slow if there are lots of them, and it does not serve any useful purpose since these are not "features" but simply fill between the outer walls.

The logic for slowing down "small features" should only apply to types of extrusion paths for which it makes sense: outer walls, support outer walls, and maybe a few others. Not for extrusion paths that are completely enclosed within and constrained by another outer structure where printing them slowly will not contribute to dimensional accuracy and surface quality of the part.

rburema, 2022-11-18:
Again, sorry for the wait.

What you say makes a lot of sense. The proposed fix (see branch above) should be very small.

rburema, 2022-11-22:
@smartavionics Up here is the problem that's happening because of this. It seems like another use-case entirely. Perhaps we should consider making a 0-th layer version of this then. Or _only_ also do it for inner walls when the layer-nr == 0. In fact, I like that last one ... I'll amend the PR. -- Should also make a front-end PR to declare this in the description then.

> If my memory serves me right, the primary reason for creating this feature was to help adhesion of small diameter features on the bottom layer so only applying it to outer walls may possibly cause prints to fail if people are relying on a slowdown to be applied to all walls and not only the outer wall. Just saying.

PR #1777 (2022-12-30): [CURA-9920] Small feature settings should only affect outer wall.
Since the stated purpose of this feture is dimensional accuracy, infill and inner walls matter a lot less. With the introduction of Arachne, small loops not in the outer walls had become more common.

should fix #1631
commit 58aef385 (2022-12-30): Small feature settings should only affect outer wall.

Since the stated purpose of this feture is dimensional accuracy, infill and inner walls matter a lot less. With the introduction of Arachne, small loops not in the outer walls had become more common.

should fix #1631
closed 2022-12-30 by jellespijker