Ultimaker/CuraEngine #1649: Overhang speed can cause artifacts from rapid flow oscillation

richfelker, 2022-04-22:
When an overhang is just on the edge of the angle to be treated with overhang speed, some segments will have the speed reduction applied while others will not. Unless acceleration is set really low, this will cause rapid oscillation of speed and thus flow, and may produce extrusion artifacts. I see at least three possible strategies to solve this:

1. Apply some sort of filter to eliminate oscillations, as is done for Arachne like splitting/merging decisions.
2. Don't use a hard cutoff for overhang but a range of angles over which the speed smoothly reduces from full speed to the reduced speed.
3. Offer a separate acceleration setting for overhangs that also applies to (some length of) the segment immediately before and after the speed-reduced segments of the path, so that the flow change can be forced to take place very gradually. In this case the printer firmware's acceleration profile will prevent the speed from increasing significantly in a vicinity of the slowed-down segments. Lower acceleration limit would probably also help overhang quality in other ways too.

rburema, 2022-12-04:
I'm going to forward these interesting FR's to Print Processing and Materials: @rijkvanmanen @pkuiper-ultimaker

rijkvanmanen, 2022-12-06:
Thank you for the FR @richfelker.
I prefer the 3rd option, especially because your reasoning does not only apply to overhang speeds, but to speed jumps in general between different line types. However, don't expect us to implement this soon.

Internal reference to this idea: PP-134