Ultimaker/CuraEngine #1650: CuraEngine wont compile with CMake

imMadsen, 2022-04-22:
**Application Version**
Latest

**Platform**
Ubuntu

**Steps to Reproduce**

**Actual Results**
Error message
```
CMake Error at CMakeLists.txt:262 (add_library):
  The target name "clipper::clipper" is reserved or not valid for certain
  CMake features, such as generator expressions, and may result in undefined
  behavior.

CMake Error at CMakeLists.txt:263 (target_include_directories):
  Cannot specify include directories for target "clipper::clipper" which is
  not built by this project.

CMake Error at CMakeLists.txt:272 (add_library):
  The target name "rapidjson::rapidjson" is reserved or not valid for certain
  CMake features, such as generator expressions, and may result in undefined
  behavior.

CMake Error at CMakeLists.txt:273 (target_include_directories):
  Cannot specify include directories for target "rapidjson::rapidjson" which
  is not built by this project.

CMake Error at CMakeLists.txt:280 (find_package):
  By not providing "Findstb.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "stb", but
  CMake did not find one.

  Could not find a package configuration file provided by "stb" with any of
  the following names:

    stbConfig.cmake
    stb-config.cmake

  Add the installation prefix of "stb" to CMAKE_PREFIX_PATH or set "stb_DIR"
  to a directory containing one of the above files.  If "stb" provides a
  separate development package or SDK, be sure it has been installed.
```

**Expected results**
Compile

jellespijker, 2022-04-22:
We're in the process of switching to Conan as our dependency manager. If you set up Conan on your computer it should be fairly simple to Download and compile all of Cura his Dependencies.
Setting up Conan is done as follows
```bash
python -m pip install conan
conan profile new default --detect
conan config install https://github.com/ultimaker/conan-config.git
```
This will install Conan and download the configuration files which should work for our dependencies. You normally only have to do this once or when we update those configuration files.

You can then install the dependencies for CuraEngine
```bash
conan install . -if build -pr:b cura_build.jinja -pr:h cura_release.jinja --build=missing
```
That command will use the conanfile.py in the root directory and download and build the required dependencies. It will then create automatic CMake Find modules in the build folder. The `-if` argument stands for install folder.
The binaries for those dependencies are either downloaded from our server, our build locally if they're not available for your compiler, architecture and OS. These are then stored in a local Conan repository and reused for this and other projects.

After this you can configure the CMake project with the auto-generated find-module from the step above. Keep this simple. Don't link to your old System Protobuf and stuff like that. The tests aren't working yet, because we were time-pressed for the 5.0 release. That is why I pass BUILD_TESTS=OFF for now.
```
cd build
cmake \
-DCMAKE_BUILD_TYPE=Release \
-DCMAKE_TOOLCHAIN_FILE=conan_toolchain.cmake \
-DBUILD_TESTS=OFF \
 ..
```

and now you can make the build
```
make -j $(nproc)
```

There should be no need to build, compile, link Protobuf and Arcus since Conan will now take care of that

jellespijker, 2022-04-22:
I will update the instructions on the README.md probably after the stable 5.0 release. That is why I pinned this issue for now

closed 2022-04-23 by imMadsen