Ultimaker/CuraEngine #1651: Failed requirement 'arcus/5.0.0-a+1@ultimaker/testing'

imMadsen, 2022-04-23:
**Application Version**
CuraEngine v5.0

**Platform**
Ubuntu 20.04.4

**Steps to Reproduce**
```
conan config install https://github.com/ultimaker/conan-config.git
conan install . -if build -pr:b cura_build.jinja -pr:h cura_release.jinja --build=missing
```

**Actual Results**
```
cmake \
> -DCMAKE_BUILD_TYPE=Release \
> -DCMAKE_TOOLCHAIN_FILE=conan_toolchain.cmake \
> -DBUILD_TESTS=OFF \
>  ..
CMake Error at /home/madsen/cmake-3.23.1-linux-x86_64/share/cmake-3.23/Modules/CMakeDetermineSystem.cmake:130 (message):  Could not find toolchain file: conan_toolchain.cmake
Call Stack (most recent call first):
  CMakeLists.txt:5 (project)

CMake Error: CMake was unable to find a build program corresponding to "Unix Makefiles".  CMAKE_MAKE_PROGRAM is not set.  You probably need to select a different build tool.
CMake Error: CMAKE_C_COMPILER not set, after EnableLanguage
CMake Error: CMAKE_CXX_COMPILER not set, after EnableLanguage
-- Configuring incomplete, errors occurred!
```

**Additional Information**
When running conan
```
conan install . -if build -pr:b cura_build.jinja -pr:h cura_release.jinja --build=missing
```
I get the following output
```
Configuration (profile_host):
[settings]
arch=x86_64
arch_build=x86_64
build_type=Release
compiler=gcc
compiler.cppstd=17
compiler.libcxx=libstdc++
compiler.version=9
os=Linux
os_build=Linux
[options]
[build_requires]
[env]
[conf]
tools.cmake.cmaketoolchain:find_package_prefer_config=True

Configuration (profile_build):
[settings]
arch=x86_64
arch_build=x86_64
build_type=Release
compiler=gcc
compiler.cppstd=17
compiler.libcxx=libstdc++
compiler.version=9
os=Linux
os_build=Linux
[options]
[build_requires]
*: cmake/3.22.0, ninja/1.10.2
[env]
[conf]
tools.cmake.cmaketoolchain:find_package_prefer_config=True

protobuf/3.17.1: Not found in local cache, looking in remotes...
protobuf/3.17.1: Trying with 'ultimaker'...
Downloading conanmanifest.txt completed [0.17k]
Downloading conanfile.py completed [13.54k]
Downloading conan_export.tgz completed [0.25k]
Decompressing conan_export.tgz completed [0.00k]
protobuf/3.17.1: Downloaded recipe revision 58cea65ea86d0d26fd1b7378c31aaf34
zlib/1.2.12: Not found in local cache, looking in remotes...
zlib/1.2.12: Trying with 'ultimaker'...
Downloading conanmanifest.txt completed [0.33k]
Downloading conanfile.py completed [5.11k]
Downloading conan_export.tgz completed [0.26k]
Decompressing conan_export.tgz completed [0.00k]
zlib/1.2.12: Downloaded recipe revision 82202701ea360c0863f1db5008067122
clipper/6.4.2: Not found in local cache, looking in remotes...
clipper/6.4.2: Trying with 'ultimaker'...
Downloading conanmanifest.txt completed [0.26k]
Downloading conanfile.py completed [3.66k]
Downloading conan_export.tgz completed [0.31k]
Decompressing conan_export.tgz completed [0.00k]
clipper/6.4.2: Downloaded recipe revision 4723a7895138c9c26926f23c46630c2a
boost/1.78.0: Not found in local cache, looking in remotes...
boost/1.78.0: Trying with 'ultimaker'...
Downloading conanmanifest.txt completed [0.69k]
Downloading conanfile.py completed [77.30k]
Downloading conan_export.tgz completed [1.38k]
Decompressing conan_export.tgz completed [0.00k]
boost/1.78.0: Downloaded recipe revision 0c3c3fe1cd8f017735fcbf4cc91d2625
gtest/1.8.1: Not found in local cache, looking in remotes...
gtest/1.8.1: Trying with 'ultimaker'...
Downloading conanmanifest.txt completed [0.24k]
Downloading conanfile.py completed [8.59k]
Downloading conan_export.tgz completed [0.29k]
Decompressing conan_export.tgz completed [0.00k]
gtest/1.8.1: Downloaded recipe revision a68c7a96d668242114b772f28235f004
rapidjson/1.1.0: Not found in local cache, looking in remotes...
rapidjson/1.1.0: Trying with 'ultimaker'...
Downloading conanmanifest.txt completed [0.10k]
Downloading conanfile.py completed [1.06k]
Downloading conan_export.tgz completed [0.25k]
Decompressing conan_export.tgz completed [0.00k]
rapidjson/1.1.0: Downloaded recipe revision 947d56f97c67f87bd1db1ea5453fbcb2
stb/20200203: Not found in local cache, looking in remotes...
stb/20200203: Trying with 'ultimaker'...
Downloading conanmanifest.txt completed [0.10k]
Downloading conanfile.py completed [1.16k]
Downloading conan_export.tgz completed [0.26k]
Decompressing conan_export.tgz completed [0.00k]
stb/20200203: Downloaded recipe revision cba8fa43f7a0ea5389896744664823c9
arcus/5.0.0-a+1@ultimaker/testing: Not found in local cache, looking in remotes...
arcus/5.0.0-a+1@ultimaker/testing: Trying with 'ultimaker'...
arcus/5.0.0-a+1@ultimaker/testing: Trying with 'conan-center'...
ERROR: Failed requirement 'arcus/5.0.0-a+1@ultimaker/testing' from 'conanfile.py (cura-build-environment/None)'
ERROR: Unable to find 'arcus/5.0.0-a+1@ultimaker/testing' in remotes
```

jellespijker, 2022-04-25:
Yes this looks like a bug, where it can't find the latest arcus.

I'm currently in the process of updating the CI/CT
I think the Arcus package has been removed accidentally.  I will take a closer look tomorrow.
What you could do for now is check out the https://github.com/Ultimaker/Curaengine/tree/CURA-9177_Fix_CI_CT

Branch this should be up to date with master and has the correct dependencies and cmake targets.

I don't expect this to be merged to master before Friday. But it will give you a working branch for now

jellespijker, 2022-04-25:
@casperlamboo and/or @Joeydelarago I think you both might have this 
`arcus/5.0.0-a+1@ultimaker/testing`
Recipe, could either of you do `conan upload Arcus -r ultimaker` that should also fix this

jellespijker, 2022-04-25:
Should be fixed now:
I have uploaded the Arcus recipe to our servers again, running a conan search query for arcus shows version 5.0.0-a+1 available again.

```powershell
C:\dev\libarcus>conan search arcus -r all
Existing package recipes:

Remote 'ultimaker':
arcus/latest@ultimaker/stable
arcus/4.9.1
arcus/5.0.0-CURA-9177-fix-CI-CT.1+45@ultimaker/cura-9177
arcus/5.0.0-PullRequest0135.43@ultimaker/testing
arcus/5.0.0-PullRequest0136.45@ultimaker/testing
arcus/5.0.0-a+1@ultimaker/testing
arcus/5.0.0-beta.1+43
arcus/5.0.0-beta.1+41
arcus/5.0.0-beta.1+19@Ultimaker/testing
arcus/5.0.0+46@ultimaker/stable
```

If you do a new `conan install ...` command it should download the recipe and build the arcus binary

jellespijker, 2022-04-25:
Let me know if this fixes your problem

closed 2022-07-13 by Joeydelarago

maruhe, 2023-01-31:
hi, I get this error again

output:
WARN: spdlog/1.10.0: requirement fmt/8.1.1 overridden by curaengine/5.3.0 to fmt/9.0.0
arcus/5.2.2: Not found in local cache, looking in remotes...
arcus/5.2.2: Trying with 'cura-ce'...
arcus/5.2.2: Trying with 'conan-center'...
ERROR: Failed requirement 'arcus/5.2.2' from 'conanfile.py (curaengine/5.3.0)'
ERROR: Unable to find 'arcus/5.2.2' in remotes

when I do >conan search arcus -r all the version 5.2.2 is not listed
...
arcus/5.2.0-beta+1@ultimaker/stable
arcus/5.2.0-beta.1+testing_3@ultimaker/testing
arcus/5.2.0-beta.1+2@ultimaker/stable
arcus/5.2.0-beta.1+1@ultimaker/stable
arcus/5.2.0-beta.1
arcus/5.2.0-beta.2+3@ultimaker/stable
arcus/5.2.1-beta.1+0@ultimaker/stable
arcus/5.3.0-alpha+testing_5@ultimaker/testing
arcus/5.3.0-alpha+testing_3@ultimaker/testing

jellespijker, 2023-01-31:
please run from the install command from `main` or add `-require-override=arcus/5.2.2@_/_` to the conan install command

maruhe, 2023-02-10:
thanks I'm using main folling this guide https://github.com/Ultimaker/CuraEngine/wiki/Building-CuraEngine-From-Source
--require-override=arcus/5.2.2@_/_ work on one machine but I walked through a forest of future errors so I tried on another windows machine
I get the same error but here even when I add --require-override=arcus/5.2.2@_/_
:-(
WARN: spdlog/1.10.0: requirement fmt/8.1.1 overridden by curaengine/5.4.0-alpha to fmt/9.0.0
arcus/5.2.2: Not found in local cache, looking in remotes...
arcus/5.2.2: Trying with 'cura-ce'...
arcus/5.2.2: Trying with 'conan-center'...
ERROR: Failed requirement 'arcus/5.2.2' from 'conanfile.py (curaengine/5.4.0-alpha)'
ERROR: Unable to find 'arcus/5.2.2' in remotes

jellespijker, 2023-02-15:
what is the output of the command
```
conan remote list
```