Ultimaker/CuraEngine #1663: CuraEngine crashes under Windows when compiled in debug mode.

ThomasRahm, 2022-05-16:
**Application Version**
5.0.0 Beta
**Platform**
Windows

I noticed some potential issues while trying to run a CuraEngine compiled under windows using the debug configuration. Both were `can't decrement vector iterator before begin` errors. 
 
1. In SkeletalTrapezoidation::generateExtraRibs in the [line 1311](https://github.com/Ultimaker/CuraEngine/blob/ac0aae947fbbead4861830e8ba65b6cd1de2098b/src/SkeletalTrapezoidation.cpp#L1311) there is an exception thrown, because in the first iteration std::prev is called with graph.edges.begin().
#1662 should fix this one.
    
2. Similarly, in `Polygons::removeSmallAreas`, the iterator `it`  can be decreased before the begin iterator when it reaches [`i--` in line 708 ](https://github.com/Ultimaker/CuraEngine/blob/ac0aae947fbbead4861830e8ba65b6cd1de2098b/src/utils/polygon.cpp#L708) in the first iteration.

I don't know if there are more cases where a crash in debug mode is caused by an iterator to be decremented before begin, as i wasn't sure how to fix the second one.

ThomasRahm, 2022-09-11:
Is there any update on this?

Piezoid, 2022-09-11:
@ThomasRahm I'm working on a patch right now :smile: 
In the meantime, you can compile with `-D_ITERATOR_DEBUG_LEVEL=0` to disable these checks.

ThomasRahm, 2022-09-19:
@Piezoid Tried it. 
This fixes the iterator issue, but there are still some "Uninitialized local variable used" issues [in comp.cpp regarding the variables `start_part_boundary_poly_idx_min` and `end_part_boundary_poly_idx_min`](https://github.com/Ultimaker/CuraEngine/blob/6734fe21c61a62a78136ef05ee79fe7f5d7abed4/src/pathPlanning/Comb.cpp#L126). 
When manually initializing the variables to 0 it works.
I can finally debug again. Thank you very much.

PR #1730 (2022-11-15): Avoid moving iterators out of range
Superseeds #1593 and #1662, resolve #1663.

As noticed in discussion of #1432, this PR introduced another case of iterators decremented before `begin()`.
While it is not an issue with release builds on platforms using pointer wrapper as iterators, it is undefined behavior after all. More practically, debug builds catches this which is annoying (iterator checking is enabled by default on mscv).

This PR resolves two cases: one in `Polygons::removeSmallAreas` and one in `SkeletalTrapezoidation::isEndOfCentral`.

I cherry-picked the `removeSmallAreas` tests fro [... cut]