Ultimaker/CuraEngine #1670: [Bug] "Infill Overlap" increases support areas.

ThomasRahm, 2022-05-26:
**Application Version**
5.0.0

**Actual Results**
When the support walls are calculated  [in AreaSupport::generateGradualSupport](https://github.com/Ultimaker/CuraEngine/blob/d002c2c331427ef2f0c29eb24541389626328244/src/support.cpp#L211), the support areas will be offset by overlap in `Infill::generateWallToolPaths`.
This causes the support area (not just the support infill area) to become larger.
While this does not matter most of the time, when the overlap is large, or the support area is small that can cause issues.
Examples:

[image]

[image]

**Expected results**

When setting the infill overlap used to calculate the walls in `AreaSupport::generateGradualSupport` to 0, the overlap is still correctly added, but the support area does not increase. 
This results in the previous examples to look like this instead:

[image]

The support infill will still overlap with the support walls:
[image]

[image]

ThomasRahm, 2022-10-04:
Are there any updates regarding this bug/corresponding pull request?

rburema, 2022-11-25:
@ThomasRahm Hi, sorry for the extreme delays on our side. Fortunately, we've restarted our efforts to get more of a handle on these recently.

PR #1671 (2022-11-25): Don't increase support area by infill overlap.
Don't increase support area by infill overlap. Fixes #1670.
commit 10f35446 (2023-01-05): Fixes #1670

Don't increase support area by infill overlap.
closed 2022-11-25 by rburema