Ultimaker/CuraEngine #1680: Trying to retrieve setting with no value given: 'adhesion_extruder_nr'

Fgodi-coriolis, 2022-06-09:
**Context**

  I was doing a very basic usage of CuraEngine 4.13.0, I try to do the same with CuraEngine 5.0.0, I use a json configuration file already included in Cura install directory and a standard Bunny.stl.

**Expected results**

No error and a gcode file, just like the following command which is working fine, using Cura 4.13 :
```
"C:\Program Files\Ultimaker Cura 4.13.1\CuraEngine.exe"
slice -j "C:\Program Files\Ultimaker Cura 4.13.1\resources\definitions\fdmprinter.def.json" 
-o "C:\Users\fgodi\Documents\gcode\curaOutput.gcode" 
-l "C:\Users\fgodi\Documents\stl\Bunny.stl"
```

**Steps to Reproduce**

Just try the corresponding command for CuraEngine 5.0.0  :
```
"C:\Program Files\Ultimaker Cura 5.0.0\CuraEngine.exe"
slice -j "C:\Program Files\Ultimaker Cura 5.0.0\share\cura\resources\definitions\fdmprinter.def.json" 
-o "C:\Users\fgodi\Documents\gcode\curaOutput.gcode"
-l "C:\Users\fgodi\Documents\stl\Bunny.stl"
```

**Actual Results**
> [ERROR] Trying to retrieve setting with no value given: 'adhesion_extruder_nr'

**Application Version**
CuraEngine 5.0.0. 

**Platform**
Windows

**Additional Information**
The full output (including the lines before the [ERROR]) is attached.
[output.txt](https://github.com/Ultimaker/CuraEngine/files/8871661/output.txt)

nallath, 2022-06-10:
Good find. We've added a bunch more specific settings for the raft, but they aren't used in the right place everywhere! 

This could actually lead to issues in some places.

shaoyidi, 2022-07-11:
I have the same error([ERROR] Trying to retrieve setting with no value given: 'adhesion_extruder_nr') when I uesed 5.1,is this error seriously?how to fix it?

nallath, 2022-07-11:
This is fixed by PR https://github.com/Ultimaker/CuraEngine/pull/1693

closed 2022-09-29 by Fgodi-coriolis