Ultimaker/CuraEngine #1684: Feature request: ordering outer walls separately wrt infill etc.

richfelker, 2022-06-22:
Presently, it's possible to get the following orders with outer walls first:

1. Infill
2. Outer walls
3. Inner walls
4. Skins

or:

1. Outer walls
2. Inner walls
3. Infill
4. Skins

But I would like to be able to achieve:

1. Outer walls
2. Infill
3. (Ideally) skins
4. Inner walls (ideally inside-out)

The order between 3 and 4 there is not really important.

The key goal here is to give the maximum possible time between laying the outer wall and putting any other material adjacent to it, so that it has time to cool and its shape is not affected by motion or extrusion imprecisions in the adjacent inner wall material. This reduced warping/corner curl at high speeds and facilitates using ludicrous levels of speed and acceleration, even above the levels where ringing becomes extreme, for inner walls without affecting outer wall quality.

I'm not sure what the right settings frame work for this would be. Maybe a single boolean to do "outer wall before everything else".

rburema, 2022-12-04:
I'm going to forward these interesting FR's to Print Processing and Materials: @rijkvanmanen @pkuiper-ultimaker

rijkvanmanen, 2022-12-06:
Thank you for your FR @richfelker.
Additionally to the options you provide, you can also swap the wall order. But still you are limited, although the order we prefer in the Print Process and Materials team is available (INFILL - OUTER - INNER - SKIN).

We have discussed the option to provide an interface to allow for any possible order in the past, but back then we decided that the value was limited.