Ultimaker/CuraEngine #1692: [ERROR] Trying to retrieve setting with no value given: 'adhesion_extruder_nr' When slice in debug model on win10/VS2019

shaoyidi, 2022-07-11:
**Application Version**

5.1(CuraEngine compiled with Conan)

**Platform**

Win10 / Visual Studio 2019

**Steps to Reproduce**

1.Follow the instruction(https://github.com/Ultimaker/CuraEngine/tree/5.1) to build in debug type;
2.Run the CuraEngine project in VS2019 Debug x64 model;
3.input the cmd:
slice -v -j "E:\Cura-5.1\resources\definitions\fdmprinter.def.json" -s support_enable=False -l "E:\CuraEngine-5.1\tests\testModel.stl" -o "E:\CuraEngine-5.1\tests\test.gcode"
4.[ERROR] Trying to retrieve setting with no value given: 'adhesion_extruder_nr'

**Actual Results**

create a test.gcode with 0kb;

**Expected results**

get a gcode file with right content

**Additional Information**

There is a output file:
[out.txt](https://github.com/Ultimaker/CuraEngine/files/9080753/out.txt)

canerd7u, 2022-07-12:
**i same error !!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!**

shaoyidi, 2022-07-13:
@canerd7u Do you solved this issue?

nallath, 2022-07-13:
This is fixed for the 5.1 final release

closed 2022-07-13 by nallath