Ultimaker/CuraEngine #1702: Feature request: split edges to honor seam position exactly

richfelker, 2022-07-28:
CuraEngine seems to pick the closest point from the set of vertices to the seam position, which might actually be the furthest point from the requested seam position, e.g. in the case of  triangle where the closest point is the midpoint of one side and the opposite point is closer than either endpoint. In less extreme situations, this causes the seam to jump around in segmentation-dependent ways, which is really ugly, and it can harm overhang performance by putting the seam on a corner that's overhanging in both directions (think: upside down pyramid type shape) instead of the requested seam position where it would only be overhanging in one direction and better supported.

This feature request is to, instead of picking the closest point from the set of vertices, pick the closest point along the entire path, possibly in the middle of a segment, and split that segment to give the exact requested seam location. In case users prefer the old behavior, this should possibly be made optional.

smartavionics, 2022-08-01:
See https://github.com/smartavionics/Cura/releases for a Cura variant that already does what you're requesting...

jellespijker, 2022-11-16:
devs see CURA-9474

jellespijker, 2024-05-30:
See #2087