Ultimaker/CuraEngine #1703: Following README instructions: ERROR: Profile not found: cura_debug.jinja

Drabbity, 2022-07-29:
**Application Version**
15.10 (main branch)

**Platform**
Ubuntu 

**Steps to Reproduce**
Installation of debug build (https://github.com/Ultimaker/CuraEngine/blob/main/README.md)

```
conan profile new default --detect

conan config install https://github.com/Ultimaker/conan-config.git -a "-b CURA-9177_Fix_CI_CD" -tf $HOME/.conan

conan install . -pr:b cura_build.jinja -pr:h cura_debug.jinja --build=missing #error
```

**Actual Results**
After I run `conan install . -pr:b cura_build.jinja -pr:h cura_debug.jinja --build=missing #error` I get an Error:
 `ERROR: Profile not found: cura_debug.jinja`

**Additional Information**
I am just following the README.md instructions. Please help. I cant get this to work.

NadavT, 2022-08-05:
Hey!
I ran into the same problem and managed to fixed it (disclaimer: it was on windows and the fix is partial).
You can run (given you have a profile, see https://github.com/Ultimaker/Cura/wiki/Building-Cura-from-Source under configure conan):
```
conan install . --build=missing
```
or
```
conan install --profile cura_build.jinja . --build=missing
```
(This one should exist)

This will create a working environment for building **in release**.
To get a debug version working you could play around with flags in the cmake, I used it to add debug prints (it's patchy but it works).

Anyway I haven't managed to make `-DCMAKE_BUILD_TYPE=Debug` work but it may help you start working until they fix this issue.

Dogm, 2022-08-09:
profiles can be spied here, I think the guys did not finish the decision
https://github.com/jellespijker/conan-config/tree/main/profiles

jellespijker, 2022-08-09:
The problem is that the readme in CuraEngine is out of date. We have a ticket on the backlog to update this CURA-9056. This will be addressed somewhere next week.

Follow the instructions in the CURA wiki for now.
https://github.com/Ultimaker/Cura/wiki/Running-Cura-from-Source

jellespijker, 2022-09-20:
Is this still an issue?

jellespijker, 2022-10-10:
I will close this issue because I believe it is resolved. If you're still experiencing problems feel free to reopen or create a new ticket

closed 2022-10-10 by jellespijker