Ultimaker/CuraEngine #1705: Feature request: different travel velocity and accel for combing vs external travel

richfelker, 2022-08-01:
High velocity travel over already-printed material can mar the surface, especially on certain materials like PETG, and high-acceleration travel over it can pass vibrations from the tool motion into not-yet-hardened plastic. But extremely high speed and acceleration for external travel are desirable to prevent oozing/stringing and long print times. It would be very nice to be able to configure these separately to get the best of both worlds.

rburema, 2022-12-04:
I'm going to forward these interesting FR's to Print Processing and Materials: @rijkvanmanen @pkuiper-ultimaker

rijkvanmanen, 2022-12-07:
Thank you for your request @richfelker. I recognize your problem, but never thought of this solution. I propose to add it to the Cura backlog @rburema. I think it is relatively easy to implement, but I am not sure about the priority.

rburema, 2022-12-08:
Placed it on the list of things to discuss with the team!

pkuiper-ultimaker, 2022-12-12:
A relative new feature is to disable travel jerk and travel acceleration, so this helps to prevent the first use case and ringing (disabled by default in Ultimaker profiles). However, this does not solve the requested fast acceleration and jerks for external moves. I think should only be done for the acceleration not the deceleration due to the ringing it can produce.
So let's indeed put it on the backlog to investigate the benifits.

richfelker, 2022-12-12:
> However, this does not solve the requested fast acceleration and jerks for external moves. I think should only be done for the acceleration not the deceleration due to the ringing it can produce.

As long as there is retraction taking place for the external moves, the retract and unretract time usually suffices for vibration to subside enough not to affect the extrusion moves.

There is no separate "deceleration" control in most printer firmware, but deceleration is not really what you want to limit anyway. It's perfectly fine to decelerate fast as long as the deceleration is separated sufficiently in time from the print moves that follow. I think what you really want to do is split off the last X *distance* of the travel, and perform that with acceleration reduced to match the acceleration of the print move that will follow. This will force the planner to drop down close to the junction velocity (usually 0) where the print will start again X distance away from the junction, and do the last little bit of deceleration slowly. The distance X can be computed from the desired settling time, which is probably something like 2-3 periods of the resonant frequence. It would be very possible to automatically incorporate the settling time afforded by unretract into this, so that the distance becomes very small or zero on retracted travel moves but nontrivial on unretracted travel.

MariMakes, 2023-01-10:
Hey @richfelker,

We've discussed the request with the team and we believe we will not work on this anytime soon and will therefore defer this issue.

Since Cura is open source, you or anyone else is more than welcome to work on this issue and create a pull request yourself.
I hope you understand.