Ultimaker/CuraEngine #1706: Build instructions are outdated and don't work

richfelker, 2022-08-02:
Apparently some new "Conan" dependency management system has been introduced, but there is no documentation in the readme for how to build with it, and following the instructions no longer works. Running cmake errors out with

```
CMake Error at CMakeLists.txt:199 (find_package):
  By not providing "Findclipper.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "clipper", but
  CMake did not find one.

  Could not find a package configuration file provided by "clipper"
  (requested version 6.4.2) with any of the following names:

    clipperConfig.cmake
    clipper-config.cmake

  Add the installation prefix of "clipper" to CMAKE_PREFIX_PATH or set
  "clipper_DIR" to a directory containing one of the above files.  If
  "clipper" provides a separate development package or SDK, be sure it has
  been installed.
```

jellespijker, 2022-08-03:
Did you look at these instructions:

https://github.com/Ultimaker/Cura/wiki/Running-Cura-from-Source

And ofcourse the readme.md in CuraEngine self?

https://github.com/Ultimaker/CuraEngine#readme

Let me know if this works for you.

jellespijker, 2022-08-03:
Also be aware that we are still working on optimizing things for: "setting up the development environment" for the next two weeks.

But don't hesitate to create a ticket if things don't work out. If you could do that in the Cura repo it will be noticed quicker by us.

I will transfer this ticket there now