Ultimaker/CuraEngine #1712: Unretract before last travel move to outer wall causes ringing

richfelker, 2022-08-19:
Without the unretract before last travel move to the outer wall, the time spend unretracting suffices to let any vibration from high travel velocity settle before the actual print moves on the outer wall begin. But with the unretract having taken place just inside the part, a high acceleration (usually perpendicular to the wall, no less) short travel move causes significant ringing, making the start of the outer wall come out a lot worse than the rest of it:

[image]

Presumably this issue is somewhat understood, as it seems to have been the motivation for adding the feature to disable travel acceleration and use the acceleration of the following wall. I do not want to do that, because high travel acceleration is key to avoiding stringing and makes a huge difference to print time. But it would make sense to always, unconditionally do this for the short travel move after unretract. I have a patch to do just that:

```
commit 09d94767e36ef9905346949611592802b304887e
Author: Rich Felker <dalias@aerifal.cx>
Date:   Thu Aug 18 03:15:38 2022 +0000

    use wall accel for move to wall after unretract

diff --git a/src/LayerPlan.cpp b/src/LayerPlan.cpp
index 122e7882b..aefcfeb22 100644
--- a/src/LayerPlan.cpp
+++ b/src/LayerPlan.cpp
@@ -1877,6 +1877,11 @@ void LayerPlan::writeGCode(GCodeExport& gcode)
                 {
                     // We need to unretract before the last travel move of the path if the next path is an outer wall.
                     gcode.writeUnretractionAndPrime();
+                    // Always use acceleration of the following extrusion so as not to introduce
+                    // ringing from high travel accel into outer wall. Normally, the unretract would
+                    // give the ringing time to settle, but here the unretract happens before the travel.
+                    if(acceleration_enabled && next_extrusion_idx < paths.size())
+                        gcode.writeTravelAcceleration(paths[next_extrusion_idx].config->getAcceleration());
                 }
                 gcode.writeTravel(path.points.back(), speed);
                 continue;
```

and it fully solves the problem for me.

It may also be desirable to try to choose a better point inside to unretract at, so that the final move is not perpendicular to the wall but shallower, maybe coming in at 45° or so. This would further reduce ringing on printers without input shaper where there's essentially always some ringing on sharp turns. But such an enhancement is beyond the scope of this bug report and not something I really have a need for myself.

richfelker, 2022-08-19:
Related: wipe moves at the end of the outer wall are also applied with the travel acceleration. This is not as bad, since in theory nothing new is being printed, but the hot nozzle vibrating to the side over the wall might transfer some of that vibration to surface artifacts. I haven't found where to change this yet, but it should probably be done too.

ky438, 2022-12-02:
any updates on this? I wish ultimaker would hire you, @richfelker :heart: 

i guess they have other problems...