Ultimaker/CuraEngine #1735: F0 speed in generated GCODE

Fgodi-coriolis, 2022-09-29:
**Application Version**
I am using the CuraEngine.exe embeded with Cura 5.1.1

**Steps to Reproduce**
I can't join stl or json file so it will be hard to reproduce but I am using the following command line :
`"Ultimaker Cura 5.1.1\CuraEngine.exe" slice -v -p -j "config.def.json" -o curaOutput.gcode -s infill_line_width=0.8 -s layer_height=0.8 -s infill_pattern=lines -s infill_line_distance=4 -s speed_infill=60 -s speed_travel=0 -s adhesion_type=none -s support_enable=False -s machine_width=100000 -s machine_depth=100000 -s machine_height=100000 -s machine_center_is_zero=true -s center_object=true -l curaInput.stl `

**Actual Results**
There are move in GCode with speed = 0
For example line 2 (actually on line 9524, layer 2 of my result) :
```
G1 X68.186 Y14.046 E157.55191
G0 F0 X67.889 Y14.085
G0 X68.226 Y15.126
G0 X68.673 Y16.087
G0 X69.663 Y17.383
G0 X70.474 Y18.063
G0 X71.901 Y18.811
G0 X73.294 Y19.132
G0 X76.562 Y19.333
G0 X76.679 Y19.611
G1 F3600 X77.687 Y19.11 E157.60837
```

**Expected results**
A GCode with appropriate speeds. What would it mean to do a move with speed  = 0 ? It would take an infinite time.

Fgodi-coriolis, 2022-09-30:
[input output.zip](https://github.com/Ultimaker/CuraEngine/files/9681947/input.output.zip)

Fgodi-coriolis, 2022-09-30:
[input output.zip](https://github.com/Ultimaker/CuraEngine/files/9681960/input.output.zip)

rburema, 2022-12-04:
Sorry it took so long to get back.

I see that you have `-s speed_travel=0` in your command-line to call the engine. While an `G0 F0` is wrong, what would you expect to happen in such a case?

Fgodi-coriolis, 2022-12-05:
Thanks a lot!
I close the ticket.

closed 2022-12-05 by Fgodi-coriolis