Ultimaker/CuraEngine #1744: Consider updating to clipper2

daid, 2022-10-12:
libclipper is one of the workhorses for CuraEngine, seeing that there is a newer version which has better performance:
http://www.angusj.com/clipper2/Docs/Changes.htm

You might get overall beter performance in Cura if this is used instead.

With love,

Me.

jellespijker, 2022-10-12:
if it isn't the man himself, the O.G.

I wasn't aware that clipper2 existed, so I will definitely look into that. Thank you for the heads up and I will bring it up with our next Cura & Cookies

rburema, 2022-10-13:
Sweet! ❤️ 
Might be a while 'till we get to it, but definitely interested.

daid, 2022-10-13:
> if it isn't the man himself, the O.G.
> 
> I wasn't aware that clipper2 existed, so I will definitely look into that. Thank you for the heads up and I will bring it up with our next Cura & Cookies

I noticed it, and directly went to check if you guys where already using it. Glad to hear Cura&Cookies is still a thing.

I expect good quality from clipper2. I only once had a bug with clipper1 and that was quickly fixed after creating a reproducible case and sending it.

jellespijker, 2022-10-13:
you got all (of the developers) very excited atleast.

I was planning on creating a Conan recipe for clipper2 in the weekend, because currently on clipper1 is in the Conan Center Index https://github.com/conan-io/conan-center-index/blob/master/recipes/clipper/all/conanfile.py). Once I made a PR for this there it would be very easy to integrate build system wise so that we can experiment with the fun stuff.

smartavionics, 2022-10-23:
Hi, I have done the experiment and now have a version of my Cura variant that uses Clipper2. Haven't extensively tested it so I don't have performance figures yet.

jellespijker, 2022-10-23:
@smartavionics was it a drop-in replacement?

smartavionics, 2022-10-23:
Unfortunately, it's not a drop-in replacement, I had to make a zillion small edits. To give you an idea, here's the diffs I made for IntPoint.h and polygon.h...

```
diff --git a/src/utils/IntPoint.h b/src/utils/IntPoint.h
index f0e4421e9..8db3336ef 100644
--- a/src/utils/IntPoint.h
+++ b/src/utils/IntPoint.h
@@ -11,7 +11,7 @@ Integer points are used to avoid floating point rounding errors, and because Cli
 #define INLINE static inline
 
 //Include Clipper to get the ClipperLib::IntPoint definition, which we reuse as Point definition.
-#include "clipper.hpp"
+#include "clipper2/clipper.h"
 #include <cmath>
 #include <functional> // for hash function object
 #include <iostream> // auto-serialization / auto-toString()
@@ -37,27 +37,29 @@ namespace cura
 {
 
 /* 64bit Points are used mostly throughout the code, these are the 2D points from ClipperLib */
-typedef ClipperLib::IntPoint Point;
+typedef Clipper2Lib::Point64 Point;
 
-#define POINT_MIN std::numeric_limits<ClipperLib::cInt>::min()
-#define POINT_MAX std::numeric_limits<ClipperLib::cInt>::max()
+#define POINT_MIN std::numeric_limits<int64_t>::min()
+#define POINT_MAX std::numeric_limits<int64_t>::max()
 
-static Point no_point(std::numeric_limits<ClipperLib::cInt>::min(), std::numeric_limits<ClipperLib::cInt>::min());
+static Point no_point(std::numeric_limits<int64_t>::min(), std::numeric_limits<int64_t>::min());
 
 /* Extra operators to make it easier to do math with the 64bit Point objects */
-INLINE Point operator-(const Point& p0) { return Point(-p0.X, -p0.Y); }
-INLINE Point operator+(const Point& p0, const Point& p1) { return Point(p0.X+p1.X, p0.Y+p1.Y); }
-INLINE Point operator-(const Point& p0, const Point& p1) { return Point(p0.X-p1.X, p0.Y-p1.Y); }
-INLINE Point operator*(const Point& p0, const coord_t i) { return Point(p0.X * i, p0.Y * i); }
+/*
+INLINE Point operator-(const Point& p0) { return Point(-p0.x, -p0.x); }
+INLINE Point operator+(const Point& p0, const Point& p1) { return Point(p0.x+p1.x, p0.y+p1.y); }
+INLINE Point operator-(const Point& p0, const Point& p1) { return Point(p0.x-p1.x, p0.y-p1.y); }
+*/
+INLINE Point operator*(const Point& p0, const coord_t i) { return Point(p0.x * i, p0.y * i); }
 template<typename T, typename = typename std::enable_if<std::is_arithmetic<T>::value, T>::type> //Use only for numeric types.
-INLINE Point operator*(const Point& p0, const T i) { return Point(p0.X * i, p0.Y * i); }
+INLINE Point operator*(const Point& p0, const T i) { return Point(p0.x * i, p0.y * i); }
 template<typename T, typename = typename std::enable_if<std::is_arithmetic<T>::value, T>::type> //Use only for numeric types.
 INLINE Point operator*(const T i, const Point& p0) { return p0 * i; }
-INLINE Point operator/(const Point& p0, const coord_t i) { return Point(p0.X/i, p0.Y/i); }
-INLINE Point operator/(const Point& p0, const Point& p1) { return Point(p0.X/p1.X, p0.Y/p1.Y); }
+INLINE Point operator/(const Point& p0, const coord_t i) { return Point(p0.x/i, p0.y/i); }
+INLINE Point operator/(const Point& p0, const Point& p1) { return Point(p0.x/p1.x, p0.y/p1.y); }
 
-INLINE Point& operator += (Point& p0, const Point& p1) { p0.X += p1.X; p0.Y += p1.Y; return p0; }
-INLINE Point& operator -= (Point& p0, const Point& p1) { p0.X -= p1.X; p0.Y -= p1.Y; return p0; }
+INLINE Point& operator += (Point& p0, const Point& p1) { p0.x += p1.x; p0.y += p1.y; return p0; }
+INLINE Point& operator -= (Point& p0, const Point& p1) { p0.x -= p1.x; p0.y -= p1.y; return p0; }
 
 /* ***** NOTE *****
    TL;DR: DO NOT implement operators *= and /= because of the default values in ClipperLib::IntPoint's constructor.
@@ -73,21 +75,21 @@ INLINE Point& operator -= (Point& p0, const Point& p1) { p0.X -= p1.X; p0.Y -= p
 
 INLINE coord_t vSize2(const Point& p0)
 {
-    return p0.X*p0.X+p0.Y*p0.Y;
+    return p0.x*p0.x+p0.y*p0.y;
 }
 INLINE float vSize2f(const Point& p0)
 {
-    return static_cast<float>(p0.X)*static_cast<float>(p0.X)+static_cast<float>(p0.Y)*static_cast<float>(p0.Y);
+    return static_cast<float>(p0.x)*static_cast<float>(p0.x)+static_cast<float>(p0.y)*static_cast<float>(p0.y);
[... 577 lines cut ...]
             {
-                ret.X = std::min(ret.X, p.X);
-                ret.Y = std::min(ret.Y, p.Y);
+                ret.x = std::min(ret.x, p.x);
+                ret.y = std::min(ret.y, p.y);
             }
         }
         return ret;
@@ -1146,12 +1130,12 @@ public:
     Point max() const
     {
         Point ret = Point(POINT_MIN, POINT_MIN);
-        for(const ClipperLib::Path& polygon : paths)
+        for(const Clipper2Lib::Path64& polygon : paths)
         {
             for(Point p : polygon)
             {
-                ret.X = std::max(ret.X, p.X);
-                ret.Y = std::max(ret.Y, p.Y);
+                ret.x = std::max(ret.x, p.x);
+                ret.y = std::max(ret.y, p.y);
             }
         }
         return ret;
```

smartavionics, 2022-10-23:
I just tried slicing a reasonably complex model and the times reported by the engine were 80 seconds for Clipper2 and 78 seconds for Clipper1. Hardly encouraging. This was using a Cura 4 based slicer, i.e. no Arachne. I'll try some different models and see if I can detect any speed benefit.

smartavionics, 2022-10-23:
Just sliced another complex model and, again, Clipper2 is very slightly slower than Clipper1. So perhaps it's not worth going to the effort of upgrading?

jellespijker, 2022-11-16:
@smartavionics I made a PR today for setting up an automatic benchmark check, might be of interest to you as well. https://github.com/Ultimaker/CuraEngine/pull/1775

loadship, 2025-07-04:
Hello, are you considering upgrading to clipper2?