Ultimaker/CuraEngine #1745: Unable to build from source in Docker container

werdnum, 2022-10-12:
**Application Version**

N/A

**Platform**

Docker on MacOS

**Qt**

N/A

**PyQt**

N/A

**Display Driver**

N/A

**Steps to Reproduce**

Dockerfile:

```
FROM ubuntu:22.04

ARG BUILDARCH

ENV DEBIAN_FRONTEND=noninteractive
RUN rm -f /etc/apt/apt.conf.d/docker-clean; echo 'Binary::apt::APT::Keep-Downloaded-Packages "true";' > /etc/apt/apt.conf.d/keep-cache
RUN --mount=type=cache,target=/var/cache/apt,sharing=locked \
  --mount=type=cache,target=/var/lib/apt,sharing=locked \
  apt update && apt install -y pkg-config autoconf automake libtool bison flex m4 libprotobuf-dev build-essential ninja-build python3 sudo python3-pip && \
  apt install -y --no-install-recommends ca-certificates libncurses-dev git slic3r openscad openscad-mcad python-is-python3 python3-venv wget
RUN bash -c 'if [[ ${BUILDARCH} = "arm64" ]]; then arch=aarch64; elif [[ ${BUILDARCH} = "amd64" ]]; then arch="x86_64"; else arch=${BUILDARCH}; fi; wget https://github.com/Kitware/CMake/releases/download/v3.24.2/cmake-3.24.2-linux-${arch}.sh -O /tmp/cmake-3.24.2-linux.sh'
RUN chmod +x /tmp/cmake-3.24.2-linux.sh && /tmp/cmake-3.24.2-linux.sh --skip-license --prefix=/usr/local
RUN git clone https://github.com/Ultimaker/CuraEngine.git /build/CuraEngine && cd /build/CuraEngine && git fetch --all --tags && git checkout -b stable tags/5.1.0
WORKDIR /build/CuraEngine
ENV CONAN_SYSREQUIRES_MODE=enabled
ENV CONAN_SYSREQUIRES_SUDO=0
RUN --mount=type=cache,target=/root/.cache,sharing=locked pip install conan --upgrade && conan config install https://github.com/ultimaker/conan-config.git && conan profile new default --detect --force && conan remote remove cura
RUN conan install . --build=missing --update
RUN cmake --preset release && cmake --build --preset release
```

**Actual Results**
```
#13 1.247 CMake Error at CMakeLists.txt:201 (find_package):
#13 1.247   By not providing "Findrapidjson.cmake" in CMAKE_MODULE_PATH this project
#13 1.247   has asked CMake to find a package configuration file provided by
#13 1.247   "rapidjson", but CMake did not find one.
#13 1.247
#13 1.247   Could not find a package configuration file provided by "rapidjson"
#13 1.247   (requested version 1.1.0) with any of the following names:
#13 1.247
#13 1.247     rapidjsonConfig.cmake
#13 1.247     rapidjson-config.cmake
#13 1.247
#13 1.247   Add the installation prefix of "rapidjson" to CMAKE_PREFIX_PATH or set
#13 1.247   "rapidjson_DIR" to a directory containing one of the above files.  If
#13 1.247   "rapidjson" provides a separate development package or SDK, be sure it has
#13 1.247   been installed.
```

**Expected results**
Should be able to build CuraEngine

**Additional Information**

Not sure what's going on here. Hacking around, i see that the following files do exist in the conan-build-release directory:

```
#13 1.478 RapidJSON-Target-release.cmake
#13 1.478 RapidJSON-release-x86_64-data.cmake
#13 1.478 RapidJSONConfig.cmake
#13 1.478 RapidJSONConfigVersion.cmake
#13 1.478 RapidJSONTargets.cmake
```

It seems like specifying rapidjson in lower case on L201 trips the system up because it wants the filename in lowercase, not mixed case... but `find_package()` has been called already a few lines earlier with the mixed case version and it looks like that defines some kind of library that should be picked up. The full output does show that the Conan toolchain is being correctly used, and some libraries like Arcus are being found, so I'm not sure what the deal is with rapidjson. It seems to be a popular library so not sure why it wouldn't just work with Conan.

I have no idea how CMake works though, so there's every chance I'm missing something obvious. But with that said, the build instructions on the wiki do not work for me.

jellespijker, 2022-10-12:
please follow the updated build instructions in the wiki https://github.com/Ultimaker/Cura/wiki/Running-Cura-from-Source

if you want to build CuraEngine without the complete Cura environment see the instructions in Readme in this repo.

jellespijker, 2022-10-12:
what happens if you work from the main or 5.2 branch for CuraEngine

there are a lot of changes in the conan scripts since 5.1

werdnum, 2022-10-12:
Thanks for getting back to me, I'll try running on the main branch and get back to you. I realise I was looking at the files at HEAD instead of the branch I was actually using. I wanted to build the latest stable version which is why I was going for 5.1 but I suppose 5.2 will be that fairly soon.

werdnum, 2022-10-12:
That seems to have done it, thanks for the tip!

werdnum, 2022-10-12:
(and by the way, LOVE what you've done here. I tried a few months ago to make a small patch to Cura/Engine and eventually just gave up because it was just too hard to get it to run)

closed 2022-10-12 by werdnum

werdnum, 2022-10-12:
I have CuraEngine successfully building in a container! I added some tips and tricks from my experience to the wiki page, hope it's helpful but if not feel free to move around/delete/whatever. Mostly I just wanted to create results for all the googling I did.

jellespijker, 2022-10-13:
great to hear, that it is now easier to setup a development version. O e of the main reasons we did this move was to lower the threshold for community developers to work with Cura and CuraEngine, hoping that we would see more quality PRs. So it makes me very happy to hear this.