Ultimaker/CuraEngine #1747: Exported GCODE files do not indicate the printer for which they were sliced

smartin015, 2022-10-15:
**Application Version**
5.1.1

**Platform**
Linux (observed; probably all platforms)

**Qt**
N/A, used from website

**PyQt**
N/A, used from website

**Display Driver**
```
05:00.0 VGA compatible controller: NVIDIA Corporation GP104 [GeForce GTX 1070] (rev a1)
	Subsystem: eVga.com. Corp. GP104 [GeForce GTX 1070]
	Kernel driver in use: nvidia
	Kernel modules: nvidiafb, nouveau, nvidia_drm, nvidia
```

**Steps to Reproduce**
Slice any STL and save it to a file, then open the file and observe the contents

**Actual Results**
The printer profile for which the file is sliced (e.g. "Creality CR-20") is not present anywhere in the file, as a comment or otherwise.

**Expected results**
The printer profile for which Cura sliced the file is indicated somewhere in the gcode file.

**Additional Information**

I'm the maintainer of the [Continuous Print Queue](https://github.com/smartin015/continuousprint) plugin for Octoprint, which has features requiring an understanding of the printer for which a gcode file is sliced (see [docs](https://smartin015.github.io/continuousprint/printer-profiles/#automatic-profile-assignment)). In particular, this is necessary for cluster/network printing when the type of printer on the network varies.

I'm intend to [add profile inference for the most popular slicers](https://github.com/smartin015/continuousprint/pull/133), but Cura has no way to examine the gcode and infer the printer to print the file with. I can contribute a PR to add this as a file comment (e.g. `; PRINTER_PROFILE:Creality CR-20`) but I wanted to file an issue first in case discussion was needed before I got started.

smartin015, 2022-11-11:
Hey, could I get a weigh-in on making this GCODE header addition so folks can identify the printer targeted by a gcode file?

smartin015, 2022-11-26:
Friendly ping?

smartin015, 2023-01-12:
Once more, with the friendliest of pings

jellespijker, 2023-01-12:
thanks for the friendly pings, I'm transferring the issue to Cura for more visibility @MariMakes fyi