Ultimaker/CuraEngine #1749: How to compile CuraEngine 5.1 with static libraries

hnnrs1092, 2022-10-19:
This is probably a simple issue to resolve but as a noob to compiling I have not been able to include the library libpolyclipping as a static library when compiling for Linux or  MacOS for CuraEngine 5.1 which is my preference and I am unclear what needs to be altered in CMakeCache.txt or CMakeLists.txt to do this. I have successfully compiled CuraEngine following the given instructions for both platforms, but CuraEngine can only then be used on the computer I compiled on due to using dynamic libraries, or ensuring the libpolyclipper library is present on the computer that is used. I have tried CMAKE_EXE_LINKER_FLAGS:STRING=-static but get an error message relating to trying to statically link a dynamic library. I'm not compiling with ARCUS, so arcus and protobuf are not an issue, but I assume statically linking these would be similar. Any help would be much appreciated.

jellespijker, 2022-10-20:
see the instructions in the Readme about  the usage of conan install.

you would have to add a `-o *:shared=False` as en extra argument for that comment. This should compile CuraEngine with all static libraries.

hnnrs1092, 2022-10-22:
Thank you for your reply and advice.

I added -o *:shared=False to the conan install command (conan install . -o *.shared=False --build=missing --update) with successful compilation but unfortunately I still end up with dynamic libraries as previous, whether compiled without arcus (set Enable support for ARCUS to OFF in CMakeLists.txt and ENABLE_ARCUS:STRING=False in CMakeCache.txt), so just libpolyclipping.so.22, or with Arcus; libArcus.so, libprotobuf.so.3.17.1.0. as shown by ldd command in Linux. Am I missing something else here? Many thanks again for your help.

hnnrs1092, 2022-11-01:
Success with Linux.  

I edited conanfile,py changing True to False for the following (I did not require Arcus):

lines 28-30
        "enable_arcus": False,
        "enable_openmp": True,
        "enable_testing": False

lines 56-61
    def configure(self):
        self.options["boost"].header_only = True
        self.options["clipper"].shared = False
        if self.options.enable_arcus:
            self.options["arcus"].shared = False
            self.options["protobuf"].shared = False

I still have issues with MacOS and rapidjson issues so unable to compile (a different issue here) and windows 10 still requires MSVS libraries which I can't see how they can be included as static libraries, so if anyone can help here it would be much appreciated.

jellespijker, 2022-11-03:
you don't have to edit the conanfile if you want to change an option, just use `-o curaengine:<option_name>=<value>` during your `conan install` or `conan create` command

closed 2022-11-03 by jellespijker