Ultimaker/CuraEngine #1754: failed to compile curaengine on win10

leelufei, 2022-10-25:
**Application Version**

cura engine 5.2.1
**Platform**

win 10, vs 2022
**Qt**

qt6
**PyQt**

**Display Driver**

**Steps to Reproduce**

https://github.com/Ultimaker/CuraEngine/wiki/Building-CuraEngine-From-Source#:~:text=cmake%20%2D%2Dpreset%20release

**Actual Results**

PS D:\cura5.1.0\CuraEngine> cmake --preset release
Preset CMake variables:

  CMAKE_BUILD_TYPE="Release"
  CMAKE_POLICY_DEFAULT_CMP0091="NEW"
  CMAKE_TOOLCHAIN_FILE:FILEPATH="D:\cura5.1.0\CuraEngine\build\generators\conan_toolchain.cmake"

-- Using Conan toolchain: D:/cura5.1.0/CuraEngine/build/generators/conan_toolchain.cmake
-- Conan toolchain: C++ Standard 20 with extensions OFF
-- The C compiler identification is MSVC 19.33.31630.0
-- The CXX compiler identification is MSVC 19.33.31630.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - failed
-- Check for working C compiler: C:/Program Files/Microsoft Visual Studio/2022/Community/VC/Tools/MSVC/14.33.31629/bin/Hostx86/x86/cl.exe
-- Check for working C compiler: C:/Program Files/Microsoft Visual Studio/2022/Community/VC/Tools/MSVC/14.33.31629/bin/Hostx86/x86/cl.exe - broken
CMake Error at C:/Program Files/Python38/Lib/site-packages/cmake/data/share/cmake-3.24/Modules/CMakeTestCCompiler.cmake:69 (message):
  The C compiler

    "C:/Program Files/Microsoft Visual Studio/2022/Community/VC/Tools/MSVC/14.33.31629/bin/Hostx86/x86/cl.exe"

  is not able to compile a simple test program.

  It fails with the following output:

    Change Dir: D:/cura5.1.0/CuraEngine/build/Release/CMakeFiles/CMakeTmp

    Run Build Command(s):C:/PROGRA~1/Python38/Scripts/ninja.exe cmTC_2c222 && [1/2] Building C object CMakeFiles\cmTC_2c222.dir\testCCompiler.c.obj
    [2/2] Linking C executable cmTC_2c222.exe
    FAILED: cmTC_2c222.exe
    cmd.exe /C "cd . && "C:\Program Files\Python38\Lib\site-packages\cmake\data\bin\cmake.exe" -E vs_link_exe --intdir=CMakeFiles\cmTC_2c222.dir --rc="C:\Program Files (x86)\Windows Kits\10\bin\10.0.19041.0\x86\rc.exe" --mt="C:\Program Files (x86)\Windows Kits\10\bin\10.0.19041.0\x86\mt.exe" --manifests  -- C:\MinGW\bin\ld.exe /nologo CMakeFiles\cmTC_2c222.dir\testCCompiler.c.obj  /out:cmTC_2c222.exe /implib:cmTC_2c222.lib /pdb:cmTC_2c222.pdb /version:0.0 /debug /INCREMENTAL /subsystem:console  kernel32.lib user32.lib gdi32.lib winspool.lib shell32.lib ole32.lib oleaut32.lib uuid.lib comdlg32.lib advapi32.lib && cd ."
    RC Pass 1: command "C:\Program Files (x86)\Windows Kits\10\bin\10.0.19041.0\x86\rc.exe /fo CMakeFiles\cmTC_2c222.dir/manifest.res CMakeFiles\cmTC_2c222.dir/manifest.rc" failed (exit code 0) with the following output:
    系统找不到指定的文件。
    ninja: build stopped: subcommand failed.

  CMake will not be able to correctly generate this project.
Call Stack (most recent call first):
  CMakeLists.txt:6 (project)

-- Configuring incomplete, errors occurred!
See also "D:/cura5.1.0/CuraEngine/build/Release/CMakeFiles/CMakeOutput.log".
See also "D:/cura5.1.0/CuraEngine/build/Release/CMakeFiles/CMakeError.log".
PS D:\cura5.1.0\CuraEngine>
**Expected results**

**Additional Information**

leelufei, 2022-10-26:
i checked the value CMAKE_LIBRARY_PATH. it was "C:/.conan/500166/1/libC:/.conan/da4fbe/1/libC:/.conan/1f4e8f/1/libC:/.conan/0012c7/1/libC:/.conan/ba6048/1/libC:/.conan/09aa61/1/libC:/.conan/16297a/1/libC:/.conan/fe81fb/1/libC:/.conan/b97841/1/libC:/.conan/0973c4/1/lib "

is it reasonable. ?

leelufei, 2022-10-27:
after configuring enviromental variables, the above issue has been  resolved. but i ran  into the following errors:

Determining if the C compiler works failed with the following output:
Change Dir: D:/001/CuraEngine/build/Release/CMakeFiles/CMakeTmp

Run Build Command(s):C:/PROGRA~1/Python38/Scripts/ninja.exe cmTC_b1f21 && [1/2] Building C object CMakeFiles\cmTC_b1f21.dir\testCCompiler.c.obj

[2/2] Linking C executable cmTC_b1f21.exe

FAILED: cmTC_b1f21.exe 

cmd.exe /C "cd . && "C:\Program Files\Python38\Lib\site-packages\cmake\data\bin\cmake.exe" -E vs_link_exe --intdir=CMakeFiles\cmTC_b1f21.dir --rc=rc --mt=CMAKE_MT-NOTFOUND --manifests  -- "C:\Program Files\Microsoft Visual Studio\2022\Community\VC\Tools\MSVC\14.31.31103\bin\HostX86\x86\link.exe" /nologo CMakeFiles\cmTC_b1f21.dir\testCCompiler.c.obj  /out:cmTC_b1f21.exe /implib:cmTC_b1f21.lib /pdb:cmTC_b1f21.pdb /version:0.0 /machine:X86  /debug /INCREMENTAL /subsystem:console  kernel32.lib user32.lib gdi32.lib winspool.lib shell32.lib ole32.lib oleaut32.lib uuid.lib comdlg32.lib advapi32.lib && cd ."

RC Pass 1: command "rc /fo CMakeFiles\cmTC_b1f21.dir/manifest.res CMakeFiles\cmTC_b1f21.dir/manifest.rc" failed (exit code 0) with the following output:
The System Cannot Find the File Specified.

ninja: build stopped: subcommand failed.

 "C:\Program Files\Microsoft Visual Studio\2022\Community\VC\Tools\MSVC\14.31.31103\bin\HostX86\x86\link.exe"  is not correct.  it should be C:\Program Files (x86)\Microsoft Visual Studio\....

any comments?

leelufei, 2022-10-27:
cl.exe and link.exe are in the same directory. why link.exe is located to wrong directory while cl.exe is located correctly ?

jellespijker, 2022-10-30:
from what I gather it looks like your mixing static compilation with dynamic msvc runtime.

did you install the Conan configuration provided by us?

leelufei, 2022-10-31:
Jelle, 
Thanks for your comment.
Now, " cmake --preset release " can succeed.
But it failed at "cmake --build --preset release"  ---- the step 7 in the following text,

**step1**
PS D:\001> pip install conan --upgrade
WARNING: Ignoring invalid distribution -yqt5 (c:\users\administrator\appdata\roaming\python\python38\site-packages)
WARNING: Ignoring invalid distribution -yqt5-sip (c:\users\administrator\appdata\roaming\python\python38\site-packages)
WARNING: Ignoring invalid distribution -yqt5 (c:\users\administrator\appdata\roaming\python\python38\site-packages)
WARNING: Ignoring invalid distribution -yqt5-sip (c:\users\administrator\appdata\roaming\python\python38\site-packages)
Requirement already satisfied: conan in c:\program files\python38\lib\site-packages (1.53.0)
Requirement already satisfied: requests<3.0.0,>=2.25 in c:\users\administrator\appdata\roaming\python\python38\site-packages (from conan) (2.27.1)
Requirement already satisfied: urllib3<1.27,>=1.26.6 in c:\users\administrator\appdata\roaming\python\python38\site-packages (from conan) (1.26.8)
Requirement already satisfied: colorama<0.5.0,>=0.3.3 in c:\users\administrator\appdata\roaming\python\python38\site-packages (from conan) (0.4.4)
Requirement already satisfied: PyYAML<=6.0,>=3.11 in c:\program files\python38\lib\site-packages (from conan) (6.0)
Requirement already satisfied: patch-ng<1.18,>=1.17.4 in c:\program files\python38\lib\site-packages (from conan) (1.17.4)
Requirement already satisfied: fasteners>=0.14.1 in c:\program files\python38\lib\site-packages (from conan) (0.17.3)
Requirement already satisfied: six<=1.16.0,>=1.10.0 in c:\program files\python38\lib\site-packages (from conan) (1.16.0)
Requirement already satisfied: node-semver==0.6.1 in c:\program files\python38\lib\site-packages (from conan) (0.6.1)
Requirement already satisfied: pygments<3.0,>=2.0 in c:\program files\python38\lib\site-packages (from conan) (2.12.0)
Requirement already satisfied: tqdm<5,>=4.28.1 in c:\program files\python38\lib\site-packages (from conan) (4.64.1)
Requirement already satisfied: python-dateutil<3,>=2.7.0 in c:\program files\python38\lib\site-packages (from conan) (2.8.2)
Requirement already satisfied: bottle<0.13,>=0.12.8 in c:\program files\python38\lib\site-packages (from conan) (0.12.23)
Requirement already satisfied: pluginbase>=0.5 in c:\program files\python38\lib\site-packages (from conan) (1.0.1)
Requirement already satisfied: PyJWT<3.0.0,>=2.4.0 in c:\program files\python38\lib\site-packages (from conan) (2.4.0)
Requirement already satisfied: Jinja2<4.0.0,>=3.0 in c:\program files\python38\lib\site-packages (from conan) (3.1.2)
Requirement already satisfied: MarkupSafe>=2.0 in c:\program files\python38\lib\site-packages (from Jinja2<4.0.0,>=3.0->conan) (2.1.1)
Requirement already satisfied: certifi>=2017.4.17 in c:\users\administrator\appdata\roaming\python\python38\site-packages (from requests<3.0.0,>=2.25->conan) (2021.10.8)
Requirement already satisfied: idna<4,>=2.5 in c:\users\administrator\appdata\roaming\python\python38\site-packages (from requests<3.0.0,>=2.25->conan) (3.3)
Requirement already satisfied: charset-normalizer~=2.0.0 in c:\users\administrator\appdata\roaming\python\python38\site-packages (from requests<3.0.0,>=2.25->conan) (2.0.12)
WARNING: Ignoring invalid distribution -yqt5 (c:\users\administrator\appdata\roaming\python\python38\site-packages)
WARNING: Ignoring invalid distribution -yqt5-sip (c:\users\administrator\appdata\roaming\python\python38\site-packages)
WARNING: Ignoring invalid distribution -yqt5 (c:\users\administrator\appdata\roaming\python\python38\site-packages)
WARNING: Ignoring invalid distribution -yqt5-sip (c:\users\administrator\appdata\roaming\python\python38\site-packages)
WARNING: Ignoring invalid distribution -yqt5 (c:\users\administrator\appdata\roaming\python\python38\site-packages)
WARNING: Ignoring invalid distribution -yqt5-sip (c:\users\administrator\appdata\roaming\python\python38\site-packages)
WARNING: Ignoring invalid distribution -yqt5 (c:\users\administrator\appdata\roaming\python\python38\site-packages)
WARNING: Ignoring invalid distribution -yqt5-sip (c:\users\administrator\appdata\roaming\python\python38\site-packages)
PS D:\001>

**step2:**
PS D:\001> conan config install https://github.com/ultimaker/conan-config.git
Trying to clone repo: https://github.com/ultimaker/conan-config.git
Repo cloned!
Copying file .gitignore to C:\Users\Administrator\.conan\.
Processing conan.conf
Copying file global.conf to C:\Users\Administrator\.conan\.
Skip README.md
Defining remotes from remotes.txt
Copying file GitHubActionsBuildEnv.py to C:\Users\Administrator\.conan\generators
Copying file GitHubActionsRunEnv.py to C:\Users\Administrator\.conan\generators
Copying file PyCharmRunEnv.py to C:\Users\Administrator\.conan\generators
Copying file VirtualPythonEnv.py to C:\Users\Administrator\.conan\generators
Copying file cura.jinja to C:\Users\Administrator\.conan\profiles
Copying file cura_build.jinja to C:\Users\Administrator\.conan\profiles
PS D:\001>

**step3**
PS D:\001> conan profile new default --detect --force
Found Visual Studio 17
Found gcc 9.2
Profile created with detected settings: C:\Users\Administrator\.conan\profiles\default
PS D:\001>

**step4**
PS D:\001> git clone https://github.com/Ultimaker/CuraEngine.git
Cloning into 'CuraEngine'...
remote: Enumerating objects: 58307, done.
remote: Counting objects: 100% (560/560), done.
[... 209 lines cut ...]
[... cut ...] r.cpp
D:\001\CuraEngine\include\infill\SierpinskiFill.h(1): warning C4819: The file contains a character that cannot be represented in the current code page (936). Save the file in Unicode format to prevent data loss
C:\Program Files (x86)\Windows Kits\10\Include\10.0.19041.0\um\windows.h(1): fatal error C1083: Cannot open include file: 'winapifamily.h': No such file or directory
[10/97] Building CXX object CMakeFiles\_CuraEngine.dir\src\LayerPlan.cpp.obj
ninja: build stopped: subcommand failed.

phedg1, 2022-10-31:
> 

I am getting this exact same error. It's not clear from your comments how you got past it, can you please elaborate?

phedg1, 2022-10-31:
> after configuring enviromental variables, the above issue has been resolved. but i ran into the following errors:
>
> Determining if the C compiler works failed with the following output: Change Dir: [...]

I am getting this exact same error, but it's not clear from your comments how you got past it. Can you elaborate?

Weirdly I'm getting this error on the debug preset. On the release preset I'm getting a completely different error in the cmake --build --preset release step (LINK : fatal error LNK1181: cannot open input file 'ws2_32.lib') but I won't go on about that here.

leelufei, 2022-11-01:
> cmake --preset release

I just re-ran all  the steps.

jellespijker, 2022-11-03:
can you please try again I synced the recipes in the `cura-ce` artifactory repository with our `cura` artifactory repository. 

maybe first clear your conan cache.
```
conan remove "*" -f
```

jellespijker, 2022-11-16:
@phedg1 fyi, for that command to work the cmake version needs to be atleast 3.23

closing this issue as completed, feel free to reopen if needed

closed 2022-11-16 by jellespijker