Ultimaker/CuraEngine #1756: is it possible to make cura engine in microsoft  visual studio?

leelufei, 2022-10-28:
anyone has the experience?

jellespijker, 2022-11-03:
On Windows our default compiler is msvc distributed by Visual Studio.

I believe Visual Studio can convert CMake projects so I would say yes it is possible.  Maybe @rburema knows in detail how to use this in Visual Studio

rburema, 2022-11-04:
On the one hand, you need to have Visual Studio installed to build for windows. On the other, right now, I'm just building from the command-line (x64 native tools, then start powershell) using conan (per the wiki), then using Visual Studio IDE for the code-completion & such.

I'm not sure it _fully_ understands the CMake output we give it... enough to do code-completion, but I don't think it understands how to build (at least not last time I tried), and run/debug. Full Visual Studio integration is something that would be nice to have.

leelufei, 2022-11-10:
I did the build of curaengine successfully few days ago in windows visual
studio community 2022.

One more question,  does the  latest version of run cura from source work
on 5.1?

rburema, 2022-11-10:
We've been developing the whole Conan-based setup starting with 5.0.0

Building Cura 5.1.x is likely to either work with the current instructions or not need much changes.

leelufei, 2022-11-11:
Thanks.

One more question, do you have instructions on how to pyinstall cura_app.py
on windows ?

rburema, 2022-11-11:
We use github workflows for that when building the official release. (See: https://github.com/Ultimaker/Cura/wiki/Build-With-GitHub-Actions )

While we've not written up a way to create an installer on a local machine, you could take a look at this github action to give you an insight for what would be needed (just skip the steps that are for mac and linux) https://github.com/Ultimaker/Cura/blob/main/.github/workflows/cura-installer.yml

Just to be sure: If you just want to run Cura and test out changes, there is no need to create an installer.

(Lastly, if you plan on publicly distributing your own version, please adhere to the (L)GPL licences and post any changed code online.)

leelufei, 2022-11-15:
Got it.
Thanks for your information.

jellespijker, 2022-11-16:
closing this as completed

closed 2022-11-16 by jellespijker