Ultimaker/CuraEngine #1764: Incorrect model calculation, the lines are not welded if if the flow is less than 100%

tca72, 2022-11-08:
Cura 5.2.0

Windows 10

**Qt**

**PyQt**

**Display Driver**

On the length of extrusion has been properly configured , I would like to setup the flow to get the proper line width.
So:
I print a cube 20x20x20 in the mode "Spiralize outer contour" to measure the line width (wall) with only one line printed in this vase mode with all normal settings. 
nozzle: 0.4
layer height = 0.2
line width = 0.44 (all all subsequents)
Flow = 100 %
I measure the wall (one line printed) that should be around 0.44mm (line width), but the average measured value is 0.67 mm. So in order to correct the line width, I modify the flow to 66% ( 0.44/0.67) *100 = 65.8%
 
After that I print again the cube and the measured value of the wall witdh is 0.45. Objective reached.

Then I decide to  untick the "spiralize outer contour" in order to print the Cube .
But first I discover that when I slice, the line appears not attached, small space between them (due to the reduction of the flow). So I was thinking probably Cura did not show the result of correction of flow in the preview, then I generate the Gcode and print it. 

But the result is the same, the lines are not attached (disjointed) , meaning the reduction of the flow is not taken into consideration when Cura is computing the line path and the preview. 

To reproduce it and see it in the preview:

1) create a cube of 20x20x20mm and import into Cura.

2) Test 1 : normal print
nozzle: 0.4
layer height = 0.2
line width = 0.44 (all all subsequents)
Flow = 100 %
Spiralize outer contour = on
CUBE_VASE_FLOW100
3) Test 2 : high flow rate reduction
nozzle: 0.4
layer height = 0.2
line width = 0.44 (all all subsequents)
Flow = 66%
Spiralize outer contour = on
CUBE_VASE_FLOW100

you can see in CUBE_VASE_FLOW100 the space between lines, Cura engine is not taking into account the flow modification in the model path.

**Actual Results**

Space between 2 lines if flow is under 100%
see CUBE_VASE_FLOW66%

**Expected results**

No space between 2 lines if flow is under 100%, Cura have to recalculate the printing path in order to take into consideration the reduction of line_size if the flow is modified (here reduced).
 

**Additional Information**

CUBE_VASE_FLOW100%
nozzle: 0.4
layer height = 0.2
line width = 0.44 (all all subsequents)
Flow = 100 %
Spiralize outer contour = on

[image]

CUBE_VASE_FLOW66%
nozzle: 0.4
layer height = 0.2
line width = 0.44 (all all subsequents)
Flow = 66%
Spiralize outer contour = on

[image]

FULL__CUBE_FLOW100%
nozzle: 0.4
layer height = 0.2
line width = 0.44 (all all subsequents)
Flow = 100 %
Spiralize outer contour = off

[image]

FULL_CUBE_FLOW66%
nozzle: 0.4
layer height = 0.2
line width = 0.44 (all all subsequents)
Flow = 66%
Spiralize outer contour = off

[image]

FULL__CUBE_FLOW100% layer 1
nozzle: 0.4
layer height = 0.2
line width = 0.44 (all all subsequents)
Flow = 100 %
Spiralize outer contour = off
here no space between lines.

[image]

CUBE_FULL_FLOW66% layer 1:
nozzle: 0.4
layer height = 0.2
line width = 0.44 (all all subsequents)
Flow = 66%
Spiralize outer contour = off
here space between lines, flow is not taken into account to recalculate the line path. 

[image]

ky438, 2022-12-02:
any update? I am seeing this problem also

tca72, 2022-12-06:
no update, I perform many test with many parameters in Cura, also change the nozzle to quality nozzle brozzl and Trianglelab, but still same behaviour