Ultimaker/CuraEngine #1770: [5.3.0-5.2.1] Building CuraEngine from source windows 10 - library machine type 'x64' conflicts with target machine type 'x86'

phedg1, 2022-11-12:
**Application Version**
5.3.0-alpha, 5.2.1 and 5.2.0

**Platform**
Windows 10 x64 22H2

**Steps to Reproduce**
I have followed the steps to build CuraEngine from source:

Open VS Community 2022 (17.4)
Open the CuraEngine directory
Open the Developer PowerShell.
rmdir build
conan remove "*" -f
pip install conan --upgrade
conan config install https://github.com/ultimaker/conan-config.git
conan profile new default --detect --force
conan remote remove cura
conan install . --build=missing --update
cmake --preset release
cmake --build --preset release

**Actual Results**
I get all the way to that last step, to build the release preset. But when it runs, and it does run for a while, at the end I get the error message:

C:\.conan\456f9dc\1\lib\polyclipping.lib : warning LNK4272: library machine type 'x64' conflicts with target machine type 'x86'
CuraEngine.exe : fatal error LNK1120: 132 unresolved externals
ninja: build stopped: subcommand failed

That error doesn't just come for polyclipping.lib, there are a bunch of other libs with the exact same warning message.

**Expected results**
I expected to be able to compile CuraEngine successfully.

**Additional Information**
I'm pretty sure this is caused by whichever link.exe and cl.exe are in my path. Using the developer powershell in visual studio no doubt causes it to load some hidden environment variables which is setting this up for me. I tried adding the location of these programs to my path so I could build this from the standalone powershell but I clearly wasn't using the right ones. I have four available, at:
C:\Program Files\Microsoft Visual Studio\2022\Community\VC\Tools\MSVC\14.34.31933\bin
They are in the subdirectories:
\Hostx64\x64
\Hostx64\x86
\Hostx86\x64
\Hostx86\x86

I have experiemented with these, trying to try all 4 to see which one works but even when rebooting, it wasn't clear that it was using the right one each time. Also, the error would swich between LNK1120 and LNK1121, saying that library machine type 'x86' conflicts with target machine type 'x64'. Whatever is happening, it kept giving basically the same error, just with the libary and target machine types various ways around.

Where is the path to the right one of these? Which subdirectory path do I use? I'm assuming that the Hostx64 is required because my computer is a x64 computer. And that maybe the nested x64, x86 dirs is to choose the target that I'm building CuraEngine for? I don't know which that should be.

I've tried this on multiple branches. Including the latest in the 5.2 branch (19 October), the latest in the main branch (12 November) and a few others. I've have also tried forcing a specific version of CuraEngine using the steps in this repos wiki. Lastly, I am always sure to remove the build directory and clear the conan cache before each new attempt. I could really use some help on this, I'm stumped.

Sorry about the truncated log, the visual studio powershell doesn't keep as much output history as the standalone powershell does.
[Partial Log 2022-11-12 CuraEngine Main Vanilla VSC2022.txt](https://github.com/Ultimaker/CuraEngine/files/9994101/Partial.Log.2022-11-12.CuraEngine.Main.Vanilla.VSC2022.txt)

@jellespijker
I'm tagging you in this because this is a developer environment issue and you seem to be the go to for that.

jellespijker, 2022-11-12:
you should use the regular Powershell for this, not the Visual Studio Powershell. Conan should be able to load the VCVARS itself.

It also seems that you're mixing 32 and 64 bit. Make sure that you start from scratch again to ensure that you're not mixing them again. **perform all the commands from the 64 regular Powershell**

1. Can you make sure you have the latest Conan version?
`pip install conan --upgrade `
You can check the [Conan Documentation]( https://docs.conan.io/en/latest/installation.html ) for the latest version and installation instructions

2. Did you try clearing the Conan Cache?
`conan remove "*" -f `

3. Can you make sure you have the latest config with this command?
`conan config install https://github.com/Ultimaker/conan-config.git`

4. Can you make sure you have a default profile?
`conan profile new default --detect --force`

5. Can you make sure you have the cura-ce Artifactory repo and not the cura Artifactory repo?
Please check `cura remote list`, this will return your Artifactory Repo
If you see cura in your list, run ` conan remote remove cura`

6. With these steps you should have a clean slate and you can follow the steps on our wiki again.
[You can find it here. ](https://github.com/Ultimaker/Cura/wiki/Running-Cura-from-Source)

If you are stuck in any way, please reach out. We are happy to help out.
The best way to do that is to provide us with your complete run-out log.
Not just the part that fails, but all the lines generated after you've given your command.
We need it see your profiles and packages.

[image]

This is a neat trick to keep our Github issues more

phedg1, 2022-11-13:
@jellespijker I did what you said. I'm using the latest commit on the 5.2 branch here (19 October).

First of all I made small script that shows I'm running the x64 powershell.
`$v = $env:PROCESSOR_ARCHITEW6432
Write-Host $v`
Then I ran through all the steps you listed and all the steps on the wiki to build CuraEngine. The step I got up to that caused the error was this one:
`cmake --preset release`

With this error:
`-- Conan toolchain: C++ Standard 20 with extensions OFF
-- The C compiler identification is unknown
-- The CXX compiler identification is unknown
CMake Error at CMakeLists.txt:6 (project):
  The CMAKE_C_COMPILER:

    cl

  is not a full path and was not found in the PATH.

  Tell CMake where to find the compiler by setting either the environment
  variable "CC" or the CMake cache entry CMAKE_C_COMPILER to the full path to
  the compiler, or to the compiler name if it is in the PATH.

CMake Error at CMakeLists.txt:6 (project):
  The CMAKE_CXX_COMPILER:

    cl

  is not a full path and was not found in the PATH.

  Tell CMake where to find the compiler by setting either the environment
  variable "CXX" or the CMake cache entry CMAKE_CXX_COMPILER to the full path
  to the compiler, or to the compiler name if it is in the PATH.`
You said "Conan should be able to load the VCVARS itself". This shows that conan did not load the vcvars. This is why I was wondering about which HostxX/xX/ path to use in the OP.

Using the regular powershell stores a lot more output lines, so I have a complete log for you here:
[Full Log 2022-11-13 Vanilla 521.txt](https://github.com/Ultimaker/CuraEngine/files/9996429/Full.Log.2022-11-13.Vanilla.521.txt)

jellespijker, 2022-11-16:
looking through the log it seems that VCVARS is set and used for the building the dependencies. See the following snippet

```
----Running------
> "C:\.conan\6e5172\1\build\generators\conanbuild.bat" && cmake --build "C:\.conan\6e5172\1\build\Release" -- -j8
-----------------
**********************************************************************
** Visual Studio 2022 Developer Command Prompt v17.4.0
** Copyright (c) 2022 Microsoft Corporation
**********************************************************************
```

it looks like the cmake command preset command fails.

can you tell me which version you use? The recommended version is 3.23 since that allows for the syntax of `cmake --preset release`

you can check the version with:
```
cmake --version
```

phedg1, 2022-11-16:
It says I'm using the right version.
```
PS C:\Users\Peter\Documents\3D_Print\_Software\Workspace> cmake --version
cmake version 3.23.2

CMake suite maintained and supported by Kitware (kitware.com/cmake).
PS C:\Users\Peter\Documents\3D_Print\_Software\Workspace>
```

phedg1, 2022-11-17:
I'm pretty sure I've followed all of the instructions. I'm still getting the same error, here is my powershell log.
[Full Log 2022-11-17 Vanilla 521.txt](https://github.com/Ultimaker/CuraEngine/files/10028456/Full.Log.2022-11-17.Vanilla.521.txt)
[CMakeOutput.log](https://github.com/Ultimaker/CuraEngine/files/10028459/CMakeOutput.log)
[CMakeError.log](https://github.com/Ultimaker/CuraEngine/files/10028460/CMakeError.log)

KrIsChUnn, 2022-11-18:
I got the same error as yours. I have tried every potential option  of "c complier" and "cxx complier" to replace th "cl" in the "generators/conan_toolchain.txt", but it seems that no complier path satisfy cmake's requirement. hope you can fix this problem. @jellespijker

parachvte, 2022-11-18:
Exactly same issue as @phedg1

Except that I'm using CMake 3.25 which is the latest one (installed via choco).
And I tried adding path `\Hostx64\x64` to my PATH to let cmake being able to get `cl.exe`.

(I have tried using 3 different Windows PC to run through the same instructions.)

jellespijker, 2022-11-24:
Looking at the logs and the different issues opened in our repositories. I find this hard to debug. Would you guys be open to a sort of live call, in which we can do some pair debugging?

I think if I can look over your shoulders I will get a better feeling what goes wrong and were it goes wrong.

phedg1, 2022-11-24:
I'm open to that. I'm at GMT+10, I can do Mondays, Fridays (today) and some weekends.

phedg1, 2022-11-26:
@jellespijker I'm free now if you're online :)

KrIsChUnn, 2022-11-28:
I fixed this problem by acident, may be useful to you guys.
Do as the wiki says until step 2 in powershell,
and do the first sentence in step 3 in powershell, too(both for release and debug)
without tapping the following "cmake ....."sentences,  open the file curaengine in vs2022(with cmake tool and msvc-v143 installed)
click "build" and "build all", and you will see curaengine.exe in cache curaengine/build/release(debug).
[image]
hope this be helpful!

phedg1, 2022-11-28:
> I fixed this problem by acident, may be useful to you guys. Do as the wiki says until step 2 in powershell, and do the first sentence in step 3 in powershell, too(both for release and debug) without [...]

This works for me! There will be a toolbar near the top of VS2022 saying something about generating cmake. Click the "Generate" text in this toolbar and you will the be able to build all.

I can finally compile Cura and CuraEngine myself for the first time since first attempting it in March.

I think this confirms that the issue is that somehow it's not using the visual studio environment variables when running those commands from powershell.

jellespijker, 2022-12-13:
Great to know it finally works for you. I will ask @rburema to check it out and maybe update document or add troubleshooting if applicable. 

@MariMakes I will close this ticket

closed 2022-12-13 by jellespijker

rburema, 2022-12-14:
I can now also confirm that the steps given for a build (almost completely) within Visual Studio given by @KrIsChUnn work great. I think this might help with 'intellisense' (basically the good autocompletion that VS normally has), since VS now 'understands' the project.

I'll add these to the wiki, thanks!