Ultimaker/CuraEngine #1791: Windows 10 debug curaengine.exe crashes while release works

burhop, 2022-12-28:
**Application Version**
5.2 

**Platform**
Windows 10

**Steps to Reproduce**
pull curaengine and cura as normal.   No need to build cura... we just need the json files.
Checkout the 5.2 branches for both
conan install . --build=missing --update
conan install . --build=missing --update -s build_type=Debug

Open the directory in VS2022 and "build all" for both configurations.  If you run from VS2022 in debug, the debugger will stop at the error.

**Actual Results**
Release works as expected.  Debug starts to work but give the error:

Expression: cannot decrement begin list iterator

void SkeletalTrapezoidation::generateExtraRibs()
{
    auto end_edge_it = --graph.edges.end(); // Don't check newly introduced edges
    for (auto edge_it = graph.edges.begin(); std::prev(edge_it) != end_edge_it; ++edge_it)
    {

**Expected results**
IDebug works like Release

**Additional Information**

**I tried using main for both curaengine and cura but the json files seem out of sync and I get a bunch of other errors before getting to this problem.  Maybe another branch I should try?**

Seems strange I'm finding this on the 5.2 branch which I assume is pretty stable.  Maybe everyone debugs on linux?  

Some compiler/linker setting issue?  VS2022?  Call stack looks like this.

 	ucrtbased.dll!00007ffbdbc0afec()	Unknown
 	ucrtbased.dll!00007ffbdbc0af37()	Unknown
 	CuraEngine.exe!std::_List_const_iterator<std::_List_val<std::_List_simple_types<cura::STHalfEdge>>>::operator--() Line 180	C++
 	CuraEngine.exe!std::_List_iterator<std::_List_val<std::_List_simple_types<cura::STHalfEdge>>>::operator--() Line 259	C++
 	CuraEngine.exe!std::advance<std::_List_iterator<std::_List_val<std::_List_simple_types<cura::STHalfEdge>>>,__int64>(std::_List_iterator<std::_List_val<std::_List_simple_types<cura::STHalfEdge>>> & _Where, __int64 _Off) Line 1146	C++
 	CuraEngine.exe!std::prev<std::_List_iterator<std::_List_val<std::_List_simple_types<cura::STHalfEdge>>>>(std::_List_iterator<std::_List_val<std::_List_simple_types<cura::STHalfEdge>>> _First, __int64 _Off) Line 1199	C++
>	CuraEngine.exe!cura::SkeletalTrapezoidation::generateExtraRibs() Line 1325	C++
 	CuraEngine.exe!cura::SkeletalTrapezoidation::generateToolpaths(std::vector<std::vector<cura::ExtrusionLine,std::allocator<cura::ExtrusionLine>>,std::allocator<std::vector<cura::ExtrusionLine,std::allocator<cura::ExtrusionLine>>>> & generated_toolpaths, bool filter_outermost_central_edges) Line 523	C++
 	CuraEngine.exe!cura::WallToolPaths::generate() Line 120	C++
 	CuraEngine.exe!cura::WallToolPaths::getToolPaths() Line 208	C++
 	CuraEngine.exe!cura::WallsComputation::generateWalls(cura::SliceLayerPart * part) Line 72	C++
 	CuraEngine.exe!cura::WallsComputation::generateWalls(cura::SliceLayer * layer) Line 89	C++
 	CuraEngine.exe!cura::FffPolygonGenerator::processWalls(cura::SliceMeshStorage & mesh, unsigned __int64 layer_nr) Line 715	C++
 	CuraEngine.exe!cura::FffPolygonGenerator::processBasicWallsSkinInfill::__l2::<lambda_1>::operator()(unsigned __int64 layer_number) Line 493	C++
 	CuraEngine.exe!cura::parallel_for::__l12::<lambda_1>::operator()(std::unique_lock<std::mutex> & th_lock) Line 175	C++
 	[External Code]	
 	CuraEngine.exe!cura::ThreadPool::work_while<`cura::ThreadPool::worker'::`2'::<lambda_1>>(std::unique_lock<std::mutex> & lock, cura::ThreadPool::worker::__l2::<lambda_1> predicate) Line 71	C++
 	CuraEngine.exe!cura::ThreadPool::worker() Line 21	C++

I'm glad to try with main if someone can suggest a working set of json files.

burhop, 2022-12-29:
This actually looks like a valid issue as Debug is testing for a condition that would be ignored in Release. It shows up with other STLs that I've tried.

`
    _List_const_iterator& operator--() noexcept {
        const auto _New_ptr = this->_Ptr->_Prev;
#if _ITERATOR_DEBUG_LEVEL == 2
        const auto _Mycont = static_cast<const _Mylist*>(this->_Getcont());
        _STL_ASSERT(_Mycont, "cannot decrement value-initialized list iterator");
        _STL_VERIFY(_New_ptr != _Mycont->_Myhead, "cannot decrement begin list iterator");
#endif // _ITERATOR_DEBUG_LEVEL == 2`

  I haven't had time to check if it is fixed in another branch or if it is actually dangerous.  Sorry I'm not more help... still ramping up on the code base.

burhop, 2022-12-30:
Looks like there has been some churn on SkeletalTrapezoidation.cpp by @jellespijker @rburema and @casperlamboo .  This may already be fixed.   If so, worth a back port to the last stable version?

rburema, 2023-01-17:
@burhop Sorry for the late response!

Yes, that issue should be fixed now. I don't think it was actually (that) dangerous. Or at least, there's almost as much danger in introducing the fixed code. We're really conservative with backporting (basically reserved for when we have complete blockers, security issues or a new first party printer profile needs to be added and we're not near releasing a new version). Which we can be, because the goal is multiple versions of Cura per year, once every two months if we're able.

burhop, 2023-01-20:
Thanks.  Yeah, I don't think it is an issue for the product, just developers working with "old" code.  In any case, I saw the fix, patched my local version, and have been rolling along fine the 3 weeks.  When the 5.3 branch is ready, I'll switch to that.

closed 2024-04-05 by wawanbreton