Ultimaker/CuraEngine #1865: Command line setting from json file have no effect when CuraEngine is called from command line

bitchmittenz, 2023-03-21:
When I try to center a part on the bed, or change the mesh_position arguments in the json file they don't seem to be doing anything.

The command line settings are enabled, and I even tried inputting the commands in the call using <-s center_object=true> but even then nothing changes. I also tried changing other parameters within the json and they update fine, however the command line settings are not being taken into effect.

Is there something I am doing wrong or something I am missing here?

**Platform:** Ubuntu, using Cura 4.13.0-1

The call I am making is: CuraEngine slice -v -p -j fdmprinter.def.json -o output_file.gcode -l input_file.stl --next"
-> I removed filepath things from the call, I have a generic fdmprinter profile I am trying to adjust

JIAHAO-ZJ, 2023-12-21:
Can you show me your json file, I don't know how to set up json file

bitchmittenz, 2023-12-29:
So after some more testing,
[fdmprinterhrdv19.def.json](https://github.com/Ultimaker/CuraEngine/files/13795184/fdmprinterhrdv19.def.json)
 it doesn't seem to take into account ANY setting I am passing in manually. Am I missing something entirely?